sandeepmistry/arduino-LoRa #1: Calling LoRa,begin(); twice

hahobson, 2016-11-24:
My radios seem to need more time establishing LoRa,begin();  So I just add the call before the if statement.  Am also getting some garbage.   Not sure what the cause is.  Using this Dorji radios:  http://www.dorji.com/products-detail.php?ProId=14

sandeepmistry, 2016-11-26:
I just tried both the sender and receiver examples with a HopeRF RFM95W module connected to a [RedBearLab Blend board](http://redbearlab.com/blend/) running at 3.3V. Both seemed fine. The Blend board is based on the same Atmel ATmega32U4 processor as the Arduino Leonardo, but has a switch to select 5V or 3.3V mode.

Could you please provide mode details on the following:
* Arduino board you are using
* Photo or more info on the wiring
* How you are supplying 3.3V to the board, some boards might not be able to provide enough current at 3.3V
* Sketch you are using
* Provide more details about "Am also getting some garbage"
* Arduino IDE version you are using

This is the first time I've tested the library with an AVR based board, so far the software in the library is working as expected for me.

hahobson, 2016-11-27:
Hi,

1.  Using the Arduino UNO sold by this company in KL .   It is most likely
a knockoff from China.

2. Please see link below and images attached.
http://www.dorji.com/docs/data/DRF1278F.pdf
https://github.com/sandeepmistry/arduino-LoRa
Connected as per the above links.

3. 3.3VDC comes from the Arduino board, but the volt meters shows almost
4VDC output.  The 12V power supply show 12.42VDC.  With two 9VDC batteries
in parallel the output power is almost 4.5VDC output.

4. Sketches are attached in the zip file

5. Adruino 1.6.13 Hourly Build 2016/10/21 5:30

It works fine up until it receives 40-70 packets then it either receives
garbage or stops receiving.

All the best,
Harry

sandeepmistry, 2016-11-27:
> 4. Sketches are attached in the zip file

Github is not showing the zip attachments, you'll have to upload manually ...

What are you using for the transmit side?

hahobson, 2016-11-27:
>    1. Sketches are attached in the zip file

Hi,
////
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  LoRa.begin(433E6);
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(1000);
}

////

All the best,
Harry

sandeepmistry, 2016-11-29:
@don @christopherdro if you get a chance, can you try the sender example with an Uno to see if the `LoRa.begin(...)` call is needed twice?

I don't have any logic level converters on hand to try out here.

sandeepmistry, 2016-12-17:
I'm closing this one for now, based on the email exchange it was either a logic level issue or something related to power. When using a Leonardo instead of the Uno knockoff there were no issues.

closed 2016-12-17 by sandeepmistry

christopherdro, 2017-01-04:
Sorry for the late response. I've haven't come across this problem either.

hahobson, 2017-01-04:
The problem went away when we replaced UNO with a board that had closer to
3.3V  SPI .