sandeepmistry/arduino-LoRa #2: problem with "set output power" in module inAir4 (and inAir9)

nicorm5, 2016-12-01:
Hello.

Exists a problem at set output power when it's used the LoRa modules inAir4 and inAir9.

The register REG_PA_CONFIG (0x09) bit 7 select PA the output pin for the switch U5 (PE4259), but the pin PA_boost (sx1276) it's not conected at switch, you must use the RFO_LF pin to transmit with this module. 

itis necesary change line 40 "#define PA_BOOST                 0x80"  for "#define PA_BOOST                 0x70" (only in this case).

I prove write a 0x7e in RegPaConfig, the range increase at 400 meters in urban enviroment.
the max power transmit isn't posible with this modules.

thank and regard, the library is very useful.

sandeepmistry, 2016-12-05:
Hi @nicorm5,

Interesting, that means the inAir's don't use RFO, but PA boost instead.

from page 109 in the data sheet ...
> 7 PaSelect
> 0 RFO pin. Output power is limited to +14 dBm.
> 1 PA_BOOST pin. Output power is limited to +20 dBm

What you you think about adding an additional optional parameter to `setTxPower` to select RFO or PA boost?

Another option is an API to select the PA output pin ...

nicorm5, 2016-12-06:
hi.
inAir's should use RFO pin, not PA boots.
in your library select PA boots with (line 40) ```PA_BOOTS 0x80``` -> ("1"000 0000)
I think it's a good idea add the optional  parameter to ```SetTxPower``` , or define the pin at the beginning program

sandeepmistry, 2017-03-12:
Closed via https://github.com/sandeepmistry/arduino-LoRa/commit/9c9914a2c46a94ad5ac04353ceb6b018254f914e.

@nicorm5 please take a look :)

closed 2017-03-12 by sandeepmistry