sandeepmistry/arduino-LoRa #3: Issue in starting LoRa

selvamaniraj, 2016-12-18:
Hello Sandeep,
      I am using ATSMAD21G18A with doorji lora sx1278. I have tried using your program that you uploaded in your github repository. I am getting an error "starting lora failed error". I am setting the frequency of 433E6 as you insisted in this link "https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md" but i am getting error. Can you help me with this issue. I have attached my source code with the screenshots and thanks in advance. 
[image]
[lora433mhz.txt](https://github.com/sandeepmistry/arduino-LoRa/files/659273/lora433mhz.txt)

sandeepmistry, 2017-01-03:
Hi @selvamaniraj,

Could you please share a photo of your wiring?

christopherdro, 2017-01-04:
I've come across this a few times as well.

If the wiring is correct, try restarting the board. This has worked for me in the past.

selvamaniraj, 2017-01-04:
Hi @sandeepmistry & christopherdro,
My wiring connections are
   Doorji lora module -      Adafruit feather board
   nss                             -      d5
   mosi                          -    mosi
   miso                          -    miso
   sck                              -    sck
   vcc                             -    vcc
   gnd                             -   gnd

sandeepmistry, 2017-01-16:
@selvamaniraj please read: https://github.com/sandeepmistry/arduino-LoRa#semtech-sx1276777879-wiring

closed 2017-01-16 by sandeepmistry

tomitoxy, 2017-02-03:
I would like refresh this theme, 
I try start LoRa bu don't have any succes.
I connect LoRa module RFM95 (Hoperf electronic) . I try with logic level converter and without him, . I have UNO, modules connect 3.3v from arduino and external source 3.3v and nothing.

any idea ? PLEASE