sandeepmistry/arduino-LoRa #4: Reading temperature register

goran-mahovlic, 2017-01-08:
Is it possible to get temperature reading from REG_TEMP register?
I have tried but it always returns 0 :(
currently Im using your library in one GPS node that I have plan to put in car, and on one node that is connected to wifi router.
https://yadi.sk/d/1L3khXql37NVZA

sandeepmistry, 2017-01-16:
@goran-mahovlic from page 92 in the data sheet it is only supported in FSK mode.

[image]

closed 2017-01-16 by sandeepmistry

XenonofArcticus, 2022-01-14:
Would an option be to briefly configure the chipset for FSK mode, sample the temperature, then switch back?

Obviously the chipset mode COULD affect the temperature, but especially when operating in a cold outdoor situation, I think the environmental impact on the temperature is much larger than the chipset functionality influence.