sandeepmistry/arduino-LoRa #5: Lora.begin() problem

tomitoxy, 2017-02-03:
I would like refresh this theme,
I try start LoRa bu don't have any succes.
I connect LoRa module RFM95 (Hoperf electronic) . I try with logic level converter and without him, . I have UNO, modules connect 3.3v from arduino and external source 3.3v and nothing.

any idea ? PLEASE

sandeepmistry, 2017-02-06:
Hi @tomitoxy,

Could you please share a photo of your wiring? I would highly recommend using a logic level converter.

tomitoxy, 2017-02-06:
[image]

from RF to logic converter  and next to arduino UNOI coonect axactly like Your diagram 
Semtech SX1276/77/78/79 wiring

VCC	3.3V
GND	GND
SCK	SCK
MISO	MISO
MOSI	MOSI
NSS	10
NRESET	9
DIO0

sandeepmistry, 2017-02-06:
Ok, the photo is not hi res enough, however if you followed the wiring in the read me things should be fine. Could you try with an external 3.3V supply, the Uno might not be able to provide enough current on the 3.3V pin.

Also, you could try to edit the library, and see what version value is received here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L79

by adding `Serial.println(version, HEX);`

> and If I have 868MHz module what I must in LoRa.begin : like this : LoRa.begin(915E6)

```
LoRa.begin(868E6)
```

tomitoxy, 2017-02-06:
This is result of cpp modyfication 👍 module version:  0
LoRa Sender
module version:  0
Starting LoRa failed!

sandeepmistry, 2017-02-07:
Definitely a wiring or hardware problem if the version is 0x00. I suggest you quadruple check everything ...

Is VCCB disconnected? I would expect it to be connected to 5V on the Arduino. Also, it looks like the 3.3V pin on the Arduino is not connected to anything, what's VCCA connected to?

tomitoxy, 2017-02-07:
No, all VCCA and VCCB is connect to power supply 
[image]

Maybe I more easy with second modules with UART/RS232 manufactured HOPE too HM-TRLR-S-868
but I don't see any suppert on github, maybe You know who make skatch  with lora rs/uart included ?

Tomek

sandeepmistry, 2017-02-08:
> Maybe I more easy with second modules with UART/RS232 manufactured HOPE too HM-TRLR-S-868 but I don't see any support on github, maybe You know who make skatch with lora rs/uart included ?

Those are not supported by this library.

What frequency is your logic level converter rated for?

Another thing to try it lowering the SPI frequency: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L50 ... maybe try 4 MHz (`4E6`) or 1 MHz (`1E6`).

orlax, 2017-02-09:
hey there, i am having the same problem of starting LoRa Failed, i have check and rechecked my wiring am i am pretty sure it is ok. after editing the library to print the module version i get: 

LoRa Sender
2
Starting LoRa failed!
 

so i am guessing that it is something else than my wiring? 

i am using an arduino uno with a logic level converter and a barebones hope rfw96m 

https://electronilab.co/tienda/modulo-transceptor-rfm96w-hoperf-lora-ultra-long-range-433-mhz/

i have change the frequency to 433E6 with no change. do you have any idea what i could try to make this work?

sandeepmistry, 2017-02-09:
@orlax looking at page 85 of the [datasheet](http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf) the version value (reg 0x42) should be 0x11, however the SX1276 modules I've tested with have a value of 0x12.

Two suggestions:
1) Try to lower the SPI frequency as I mentioned in https://github.com/sandeepmistry/arduino-LoRa/issues/5#issuecomment-278191397
2) Remove the version check from the .cpp, and see if things continue to work.

orlax, 2017-02-09:
@sandeepmistry thank you for the suggestion, setting SPI frequency to 1E6 and using the wiring layout from here: https://things4u.github.io/HardwareGuide/Arduino/Mini-Sensor-HTU21/mini-lora.html  solved my issue althougth i dont fully understand the difference between connecting only DIO0 and connecting them all?

sandeepmistry, 2017-02-10:
@orlax thanks for trying the out!

Did you try any other SPI frequencies, like `8E6` or `4E6`? Let me know, then I can edit the library to the appropriate SPI frequency for AVR boards.

Also, what logic level converter are you using?

sandeepmistry, 2017-02-27:
@tomitoxy @orlax ping ...

orlax, 2017-02-27:
Hey there, i tried 10E6 and 4E6 only one that work was 1E6

the module i was using was
RFM96W HopeRF
i dont have the reference for the logic converter, but will find out and
post.

sandeepmistry, 2017-02-27:
@orlax 

> i dont have the reference for the logic converter, but will find out and post.

Thanks, that would be great!

I'll retest with my Dragino shield and Uno, but from what I remember the default SPI frequency was fine. Maybe each logic level converter has it's own max speed. I think it should be fine to add a `LoRa.setSPIFrequency(frequency)` API to this lib.

sandeepmistry, 2017-03-12:
I've added the `LoRa.setSPIFrequency(frequency)` API in https://github.com/sandeepmistry/arduino-LoRa/commit/ee096727594b7121c4df02927646107d5c360039. **Note:** the RadioHead Library uses a default SPI frequency of 1 MHz.

closed 2017-03-12 by sandeepmistry

aniketmad, 2018-02-20:
hi, we are using sx1276 with 868mhz and arduino uno board

we got sender senior monitor as:
SLoRa Sender
Sending packet: Sending packet: Sending packet: 

but at LoRa receiver
LoRa Receiver
what should i do to receive the packets??

Apeksha9, 2019-10-18:
Hello, 
I have a LoRa sender connected to a Nano and the LoRa receiver connected to a UNO.  The issue is that the sender is not getting initialized. The LoRas that I am using are LoRa-02, SX1278. 

**PIN CONNECTIONS OF THE SENDER:
/ Nano LoRA
// D2   DIO0
// D9   RST
// D10  NSS
// D11  MOSI
// D12  MISO
// D13  SCK**

**PIN CONNECTIONS OF THE RECEIVER:
// UNO        LoRA
// 3V3        3V3
// GND        GND
// D2         DIO0
// D9         RESET
// D13        SCK
// D12        MISO
// D11        MOSI
// D10        NSS**

How can I proceed with this problem?

Thanking you, 
Apeksha Nayak

EsKliwoN, 2022-01-24:
> Definitely a wiring or hardware problem if the version is 0x00. I suggest you quadruple check everything ...
> 
> Is VCCB disconnected? I would expect it to be connected to 5V on the Arduino. Also, it looks like the 3.3V pin on the Arduino is not connected to anything, what's VCCA connected to?

What if my problem the REG_VERSION is FF?

HefnySco, 2023-04-01:
> @orlax looking at page 85 of the [datasheet](http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf) the version value (reg 0x42) should be 0x11, however the SX1276 modules I've tested with have a value of 0x12.
> 
> Two suggestions:
> 
> 1. Try to lower the SPI frequency as I mentioned in [Lora.begin() problem #5 (comment)](https://github.com/sandeepmistry/arduino-LoRa/issues/5#issuecomment-278191397)
> 2. Remove the version check from the .cpp, and see if things continue to work.

This is probably 1278 module . I have this module and this is its reply from the datasheet.
[image]