sandeepmistry/arduino-LoRa #8: Issue of transreception

ashvindv, 2017-02-23:
Here i am wrking on project of sending and receiving gps data using 2 lora sx1276. Here i am able to either send or receive the data. But i need both transreception at same time and also in continous transreception form i.e., both must be sending and receiving gps data the whole time when its switched on. Can you please help me out. Thanks in advance

sandeepmistry, 2017-02-27:
Hey @ashvindv,

The LoRa protocol is only half duplex, so sending and receiving data simultaneously is not supported.

closed 2017-02-27 by sandeepmistry

ashvindv, 2017-03-08:
Dear sandeep,
If it is a half duplex system. Can i at least send the gps data
continuously to the receiving lora?

Thanks in advance

sandeepmistry, 2017-03-12:
@ashvindv yes.

ashvindv, 2017-03-13:
Dear sandeepmistry. Can i get the code to continously send gps data to
another lora?

Thans in advance