sandeepmistry/arduino-LoRa #9: Initialization failed

JBravo2323, 2017-03-05:
I set up two Lora radios using the RH_RF95 library. I used two Arduino MKR1000 and two Adafruit RFM95W radios. My system works but I found your Library much cleaner and easier to use.  But when I installed it and set the pins  LoRa.setPins(7,2,1); which is the same setting I used for the other library and run the program I get: "Starting LoRa failed!".
Can you help me debug/determine what is wrong?
Thanks

JBravo2323, 2017-03-05:
[image]
I am pretty sure my wiring is OK since the radios work with the other Library (RH_RF95) but just in case here is a high res photo of one of the radios (both are identical)
Thanks

sandeepmistry, 2017-03-12:
Hey @JBravo2323,

Just to confirm

* 7 is connected to NSS
* 2 is connected to NRESET
* 1 DIO0

It's a bit tricky to look at the photo as the MKR1000 has the pin labels on the bottom.

sandeepmistry, 2017-03-12:
Also, could you please add a `Serial.println(version, HEX)` statement here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L79 and let us know what the reported version is.

sandeepmistry, 2017-04-17:
Closing this for now due to lack of feedback.

closed 2017-04-17 by sandeepmistry