sandeepmistry/arduino-LoRa #10: The Lora.endPacket() gets stuck during the second pass

JBravo2323, 2017-03-11:
Hello Sandeep
I added these Serial.println for debugging in the Library:
{
  Serial.println("LoRaClass::endPacket: before   writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);");
  // put in TX mode
  writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);

  Serial.println("LoRaClass::endPacket: before while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);");
  // wait for TX done
  while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);

  // clear IRQ's
  Serial.println("LoRaClass::endPacket before  writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);");
  writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);

  return 1;
}

*************When I run the program this is what I get: ***********************

LoRa Sender
7
From the Library Definintion version = 9
Sending packet: 0
LoRaClass::endPacket: before   writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);
LoRaClass::endPacket: before while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);
LoRaClass::endPacket before  writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
Sending packet: 1
LoRaClass::endPacket: before   writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_TX);
LoRaClass::endPacket: before while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);

So it appeared it never leaves this while
while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);
Regards

sandeepmistry, 2017-03-12:
Hey @JBravo2323,

What sketch are you using? Also, what hardware?

JBravo2323, 2017-03-16:
Lora Sender and Lora Receiver are the sketches
The hardware is the Adafruit RFM95W LoRa Radio Transceiver Breakout - 868 or 915 MHz
Thanks

sandeepmistry, 2017-03-16:
@JBravo2323 what Arduino board are you connecting to the  Adafruit RFM95W LoRa Radio Transceiver Breakout?

JBravo2323, 2017-03-17:
MKR1000

Sent from Lotus Traveler

sandeepmistry, 2017-03-17:
Could you please update to the latest release and try to lower the SPI frequency via: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

Try values of: 8e6, 4e6, and 1e6.

sandeepmistry, 2017-04-17:
Closing this for now due to lack of feedback.

closed 2017-04-17 by sandeepmistry

vinniroy, 2018-11-18:
Hey @sandeepmistry 

sender code : 
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  LoRa.setSPI(SPI);
  LoRa.setSPIFrequency(4e6);
  LoRa.setSpreadingFactor(10);
  LoRa.setSignalBandwidth(42.5E3);
  LoRa.crc();
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  
  LoRa.print("hello ");
 // Serial.print("hello ");
  LoRa.print(counter);
//  Serial.println(counter);

 // if(counter==1)
  LoRa.endPacket();
  //Serial.print("asdfgh");
  
  Serial.print("Lora Send: ");
  Serial.println(counter);
  counter++;

  delay(2000);
}

Serial Output : 
LoRa Sender
Sending packet: 0
Lora Send: 0
Sending packet: 1

Problem : I have the same problem. Its not sending packet 1 after sending packet 0 due to some problem in the statement Lora.endPacket();

Kindly help asap. I already tried setting the SPI frequency to 1e6,4e6 and 8e6. It didn't worked. 
Board used: Arduino Uno
Chip used : Ra-01_V1.4