sandeepmistry/arduino-LoRa #11: REG_SYNC_WORD is set to 0x39 datasheet calls for 0x28..0x2F

kc2uez, 2017-03-14:
Thanks for making this lib available to everyone. It makes working with these modules more human readable. 
I am working on a little payload computer for a high altitude balloon. I am using your lora library to communicate with the Pi in the sky gateway from here: https://github.com/PiInTheSky/lora-gateway
I  keep getting crc error, but everything seems to match except for syncword. While looking more into this I notice that REG_SYNC_WORD may be set incorrectly to 0x39. Also REG_MODEM_3 (0x26) which if bit 3 (0x04) is set it greatly improve the reception for low data rate and noisy environments and long distances, but it is not set.

I am using the little module from 915 adafruit module which has the RFM96 IC although they said it should have the 95w. Anyways I am reading the datasheet from: http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf (page 93 has the sync word registers listed).

sandeepmistry, 2017-03-16:
Hey @kc2uez,

Page 93 appears to be for FSK/OOK key mode not LoRa.

The registers are based on the Semtech data sheet attached (page 114):

[image]

[sx1276_77_78_79.pdf](https://github.com/sandeepmistry/arduino-LoRa/files/846363/sx1276_77_78_79.pdf)

Would you be able to shared the sketch you are testing with? The examples don't have the CRC enabled.

Thanks for the tips on `REG_MODEM_3`, just to confirm, are you talking about the`AgcAutoOn` or `LowDataRateOptimize` field?

[image]

sandeepmistry, 2017-03-16:
After some testing, bit 2 (0x04) for AgcAutoOn is the one (I was confused by your bit 3 comment, but it just looks like a typo). I've added this in https://github.com/sandeepmistry/arduino-LoRa/commit/c1eb2b70f4c6322708ea1149ff86120fa92f2bb3. Thanks again for the tip!

kc2uez, 2017-03-16:
Hi, 
Yes, I was looking at the wrong register set, I am new to lora and it was the first time I heard of SyncWord.

This is the code I am working on: http://pastebin.com/fTYwAc06
I am using the adafruit RFM95w breakout board and their arduino datalogger from them too, it has a GPS module and SD card. I should be getting the drogino lora shield soon.

Anyways, I was comparing the configuration registers 0x1D after I did my pre-configuration and the output for 1D is 0x8 when I call  LoRa.dumpRegisters(Serial);. I think 
  byte signalBandwidth = 125E3;
  byte codingRateDenominator = 8;
Should have produced 1D x 78 according to the page 112 of the datasheet you linked. I looked at void LoRaClass::setSignalBandwidth(long sbw) and it looks ok and void LoRaClass::setCodingRate4(int denominator) is masking correctly I think, but for some reason when I read the values, bandwidth is not set. Am I doing something wrong?

sandeepmistry, 2017-03-16:
So I think the problem is the `byte` type only holds value between 0 - 255. Try `long signalBandwidth = 125E3;` instead.

I'm guessing the `byte signalBandwidth = 125E3;` value truncates the value to 0.

> I keep getting crc error, but everything seems to match except for syncword.

Based on the sketch you provided, `LoRa.noCrc();` turns the CRC off, it seems the Pi in the sky gateway code by default enables the CRC.

I'm closing this for now, but please let me know how the above suggestions work out.

closed 2017-03-16 by sandeepmistry