sandeepmistry/arduino-LoRa #13: How to receive and send at the same time?

846354866, 2017-04-01:
Hi, how to receive and send at the same time?
I put the example in the reception and send written together, but still can not get the results I want.
I need your help.
Thank you!

anthonywebb, 2017-04-06:
I am also interested in an example of sending a "ping" and waiting for a "pong" back.  Sending and receiving in the same sketch would be a common use case.  Most messages would want an ack of some sort.

sandeepmistry, 2017-04-10:
@846354866 could you please share the sketch you were trying?

846354866, 2017-04-10:
@sandeepmistry I read the SX1278 data manual found only half duplex, I think to send and receive before switching to enter the 'IDLE' mode.

sandeepmistry, 2017-04-10:
@846354866 that's correct.

sandeepmistry, 2017-05-28:
Please see the new duplex examples from @tigoe in:

* https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaDuplex
* https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaDuplexCallback

closed 2017-05-28 by sandeepmistry