sandeepmistry/arduino-LoRa #14: When to run setSignalBandwidth and setCodingRate4

jeskata, 2017-04-20:
Are these supposed to be run before or after LoRa.begin()? It seems radio does not work at all if run after, but do they come in effect if run before?

sandeepmistry, 2017-04-23:
@jeskata 

> Are these supposed to be run before or after LoRa.begin()?

After `LoRa.begin()`.

Could you please share the sketches you are using to test?

> It seems radio does not work at all if run after, but do they come in effect if run before?

Could you please expand on what "radio does not work at all if run after" means?

jeskata, 2017-04-25:
Turned out it's only setSignalBandwidth that causes problems;
I have basically used the LoraSender/Receiver examples on SX1278 (433 MHz), this is the setup function:
`void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  LoRa.setSPIFrequency(1E6);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setTxPower(17); //2-17
  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setCodingRate4(8); //5-8
  LoRa.setSignalBandwidth(7.8E3); 
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.
}`
with this, no data is received by LoraReceiver.

while this works:
`void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");
  
  LoRa.setSPIFrequency(1E6);
  LoRa.setSignalBandwidth(7.8E3); 
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  
  LoRa.setTxPower(17); //2-17
  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setCodingRate4(8); //5-8  
}`

I did copy the setup function to both sides so it is identical.

jeskata, 2017-04-25:
Actually it seems the higher speeds work better and too high TxPower caused packets to get garbled, but there is still some issues; this is the entire Receiver sketch:
```
#include <SPI.h>
#include <LoRa.h>

#define ACK 0xB5
int recRSSI;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  LoRa.setSPIFrequency(1E6);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setTxPower(2); //2-17
  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setCodingRate4(8); //5-8
  LoRa.setSignalBandwidth(31.25E3);
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    recRSSI = LoRa.packetRssi();

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(recRSSI);

    delay(10);
    LoRa.beginPacket();
    LoRa.write(ACK);
    LoRa.write((int8_t) recRSSI);
    LoRa.endPacket();
    Serial.println("Sent ack");
  }
}
```

And the Sender:
```
#include <SPI.h>
#include <LoRa.h>

#define ACK 0xB5
#define WAITACK 10000

int counter = 0;
unsigned long sendTime = 0;
bool ackedpacket;
int lostPackets = 0;
int packetSize;
int8_t recRSSI;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");
  
  LoRa.setSPIFrequency(1E6);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  
  LoRa.setTxPower(2); //2-17
  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setCodingRate4(8); //5-8  
  LoRa.setSignalBandwidth(31.25E3); 
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.
}

void loop() {
  ackedpacket = 0;
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  Serial.println("Written");
  LoRa.endPacket();
  Serial.println("Sent");

  sendTime = millis();
  do
  {
    packetSize = LoRa.parsePacket();
    if (packetSize) {
      Serial.println("Received packet");
      // read packet
      while (LoRa.available()) {
        if(LoRa.read() == ACK){
          Serial.print("Received ack");
          recRSSI = LoRa.read();
          
          // print RSSI of packet
          Serial.print(" with RSSI ");
          Serial.print(LoRa.packetRssi());
          Serial.print(", SNR ");
          Serial.print(LoRa.packetSnr());
          Serial.print(", remote RSSI ");
          Serial.println(recRSSI);
          
          ackedpacket = true;
        }
      }
    }
  } while((millis() - sendTime < WAITACK)&&(!ackedpacket));
  
  if(!ackedpacket) lostPackets++;

  Serial.print("Packets lost: ");
  Serial.println(lostPackets);
  
  counter++;

  delay(5000);
}
```

At this speed it still works but at lower speeds the Sender never receives ACK. I have tried increasing WAITACK (100000 = 100s) as obviously the sending is very slow.
At 15.6E3 it seems the packets are getting garbled also at lowest power. Getting something like this (first 20cm, then 2m distance):

```
LoRa Receiver
Received packet 'hhla�4' with RSSI -20
Sent ack
Received packet 'hello)0' with RSSI -20
Sent ack
Received packet 'hello)6' with RSSI -19
Sent ack
Received packet 'hello$0' with RSSI -19
Sent ack
Received packet 'hhllb	4' with RSSI -19
Sent ack
Received packet 'h�la 4' with RSSI -90
Sent ack

```

sandeepmistry, 2017-05-08:
@jeskata I don't think it's an issue with the library it's self per say, there's probably a combination of settings on the SX1276 that don't work that well for your environment.

Thoughts?

Btw, a TX power of 2 is pretty low.

sandeepmistry, 2017-05-28:
Closing for now due to lack of activity.

closed 2017-05-28 by sandeepmistry

sapana13, 2018-02-25:
can you provide code to connect NEO 6M with rf98 ..