sandeepmistry/arduino-LoRa #20: Preamble issue

fxsh1301, 2017-05-11:
I'm trying to use one of 1276 module be a gateway and collect data from other nodes.
What I know about Lora is using the way like TDMA to avoid the collision.
Distribute the time and polling  all nodes by gateway to get the data.
The nodes will sleep and wake up periodically and listen the preamble to judge if should reply or not.
Does the parameter of preamble need to set in both of gateway and nodes?
Is there any example? Thanks.

sandeepmistry, 2017-05-15:
> Does the parameter of preamble need to set in both of gateway and nodes?

You can set the preamble length on both the send and received using: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#preamble-length

There is no example of this currently.

> What I know about Lora is using the way like TDMA to avoid the collision.

LoRa doesn't have anything like this, LoRaWAN does include something like this. If you're using pure LoRa you'll have to build something yourself to avoid the collisions.

closed 2017-05-15 by sandeepmistry

fxsh1301, 2017-05-15:
I think i Confused the LoRa and LoRaWAN, thanks for explanation!
Forgive my rudeness, I'm trying to use Raspeberry Pi to be the gateway, but
I can't find any library for Pi using C. Is there any information about it
or possible transform arduino-Lora to C code and running on Raspberry ?
Thanks a lot.

Best regards.

sandeepmistry, 2017-05-15:
Hey @fxsh1301,

I haven't used anything in C. However, if you don't mind Node.js/JavaScript you mind find this interesting: https://github.com/sandeepmistry/node-sx127x

fxsh1301, 2017-05-17:
> sandeepmistry/node-sx127x

Thanks for sharing!
But when I wired the module with Raspberry.
The SPI test result showed it didn't work! Even I enabled it.
[image: 內置圖片 1]
The test code is from here
<https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting>.
The result showed that is not enable SPI or error wiring.
Did you have the same problem ?
The OS is jessie with firmware update.Raspberry Pi3
[image: 內置圖片 2]

Thanks. Best regards.

Yi-Siang Ciou, B.Eng.

Taiwan, R.O.C.

E-mail: fxsh13002@gmail.com

Line ID :fxsh1301