sandeepmistry/arduino-LoRa #25: Starting Lora failed

Momshad-Dinury, 2017-06-01:
I am using SX1278 for communication. I am using two ARDUINO board MEGA and UNO.
Using two SX1278 module to send data back and forth. 
I included Lora library and using the Sender and receiver example.
But
at the
if (!LoRa.begin(434E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
I am getting this message in the Serial Monitor That Starting Lora failed. I changed the frequency too.

EmbeddedAndMore, 2017-06-09:
the problem is in line 491 of LoRa.cpp (readRegister method)
in fact the dummy byte for SPI is chosen wrong (0x00)  because it's a register address.
in order to make it work for my sx1287 chip i changed it to **0xFE** and now it works.

return singleTransfer(address & 0x7f, `0xFE);`

hope it help you as well.

sandeepmistry, 2017-06-18:
@EmbeddedAndMore interesting, could you please submit a pull request for your recommended change.

@Momshad-Dinury did you get a chance to try @EmbeddedAndMore's suggestion?

Momshad-Dinury, 2017-06-18:
Actually my problem was solved. I selected Lora normal mode. And now its working

Momshad-Dinury, 2017-07-05:
@sandeepmistry Nope i didn't.
@EmbeddedAndMore  Are you asking me to pull a request?

ehsanbalouch, 2017-07-25:
Hi,
I'm using 410-525MHz AI Thinker Ra-01 module with Arduino UNO 3 (TX) and Nano 3 (RX).
I changed the Line 491 in LoRa.cpp , it started recognising the modules but it's not receiving data.
I am using LoRaSender.ino and LoRaReceiver.ino
Can you please help to make it working?

sandeepmistry, 2017-07-31:
Closing this as resolved for now.

@ehsanbalouch please open a new issue and open a pull request for the change you made to get things working.

closed 2017-07-31 by sandeepmistry

ehsanbalouch, 2017-07-31:
My issue is yet not resolved. Case should not be closed. I need Arduino Library and Code to make LoRa SX1278 working.

CHAWKI94, 2018-05-16:
am using a Adafruits feather32u4, i wrote this code and i got a problem is LoRa.Begin() function,
------------------------
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
---------------------------------
on the serial monitor i got this:

LoRa Sender
Starting LoRa failed!

any suggestion please?

ehsanbalouch, 2018-05-16:
It shows that LoRa module is not responding to Arduino.
You have to check the voltage to LoRa module, some are 3.3v so you have to
be careful. If you supplied 5v then module might have been damaged.
If module is 5v then it can communicate with Arduino board but for 3.3v
module you have to use level shifter.
Also check every connection thoroughly to ensure it is correct.

Regards,
Ehsan Balouch.

shahid44, 2018-07-22:
I have arduino uno and esp8266 NodeMCU. I want to communicate them with each other with Ra-01 Ai-Thinker 433 MHz. Please help me with codes, connections diagram and libraries.

ehsanbalouch, 2018-07-29:
Extremely sorry for late reply.

Please accept my apologies that I haven't done this cascading, I really
regret for not being able to help you out on this.
I just cascaded two UNO with LoRa 433 MHz.

Regards,
Ehsan Balouch.

shahid44, 2018-08-04:
Now I have tried with two UNO with Ai-Thinker Ra-01 LoRa 433 MHz, I have also corrected line 491 but LoRa sender not sending any packet and receiver not receiving 'Hello' packet.

pazmat, 2019-09-22:
I am stuck with the same issue. I have tried quite a few things but failing again and again. My LoRa module doesn't have any pin specified as ss, reset or dio how do i do this. Check the attachment to see the module i am using and help. I had made it work wth softserial though but now after a few days even softwareserial is not working.. I need help urgently. Thanks
[image]

AwaisSiawa, 2020-03-11:
CAN YOU PLEASE HELP?
I put below code for sender LORA but it said lora sender failed. not sure why. connection is correct. i changed frequency like 433, 868, 915, 434 but still the same error.
please let me know asap

#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(434E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

ehsanbalouch, 2020-03-19:
Please mention which hardware are you using. There are faults found in new
modules too. Mention its details and also photo if possible.

(Sorry for late reply because I don't check email too frequently).

Regards,
Ehsan Balouch.

IoTThinks, 2020-03-20:
@AwaisSiawa : Why don't you create new thread with hardware photo...?

alfantamamy, 2020-04-07:
I got the same problems. I tried connec- ting the Module LoRa sx1278 to Arduino Mega 2560 as sender and receiver. But didn't works ☹️. 

This is my schema, 
Pin 50 - miso
Pin 51 - mosi
Pin 52 - sck
Pin 53 - nss or ss
Pin 11 - rst
Pin  9 - dio0

Please help me, thank you.

ehsanbalouch, 2020-04-07:
I understand that these modules are not working easily.
First of all your hardware should be good one and then your code must be
simplest one.

Below is my tested code and works properly. When you run this code, you
have to enter T in the Serial Monitor to send the packet.

// lora_master.pde

// Caution:lora modeule  can only work under 3.3V
// please make sure the supply of you board is under 3.3V
// 5v supply will destroy lora module!!

// This code runs in master mode and  works with lora_slave.pde
// Flow:receive "T" from serial->send a packet->wait for reply
// data of packet is " Santech Test 01 TX0 "

#include <LORA.h>
#include <SPI.h>
#include <SoftwareSerial.h>
LORA lora;
unsigned char tx_buf[]={"Santech Test 01 TX0"};
unsigned char val;
unsigned char flag=1;    //  flag of rx mode
unsigned char rx_len;
unsigned char rx_buf[20];

void setup() {
  Serial.begin(9600);
  if(!lora.init())
  {
     Serial.println("Init fail!");
  }
flag=0;
}
void loop()
{
//    val=Serial.read();  // please make sure serial is OK before running
this code
//if (flag==0){val='T';}

//if(val=='T')    // tx a packet if receive "T"
if (flag==0)
{
       lora.txPacket(tx_buf,sizeof(tx_buf));
       lora.rxInit();    // turn to rx mode to wait for reply of slave
       ...
       flag=1;           // enable rx flag
       Serial.print("tx: ");
       Serial.print(millis());
       Serial.print("// ");
    }
  if(flag==1)
    {
        if(lora.waitIrq(LR_RXDONE_MASK))    // wait for RXDONE interrupt
        {
            flag=0;                          // clear rx flag
            lora.clrInterrupt();
            rx_len=lora.rxPacket(rx_buf);    // read rx data
            Serial.write(rx_buf,rx_len);    // print out by serial
            Serial.println();
            lora.enterStbyMode();
        }
    }

delay(700);
}

Regards,
Ehsan Balouch

IoTThinks, 2020-04-07:
> I got the same problems. I tried connec- ting the Module LoRa sx1278 to Arduino Mega 2560 as sender and receiver. But didn't works .
> 
> This is my schema,
> Pin 50 - miso
> Pin 51 - mosi
> Pin 52 - sck
> Pin 53 - nss or ss
> Pin 11 - rst
> Pin 9 - dio0
> 
> Please help me, thank you.

Show us your physical wiring photo.
Show you simplest LoRa code.
And define "not working"

Suddenly, a lot of people raise issue with Arduino Mega, Strange.

alfantamamy, 2020-04-07:
> I understand that these modules are not working easily. First of all your hardware should be good one and then your code must be simplest one. Below is my tested code and works properly. When you [...]

I'll try it, thanks before

alfantamamy, 2020-04-07:
> Show us your physical wiring photo.
> Show you simplest LoRa code.
> And define "not working"
>
> Suddenly, a lot of people raise issue with Arduino Mega, Strange.

alfantamamy, 2020-04-07:
This is code for the Receiver.

#include <SPI.h>
#include <LoRa.h>

#define rst 11
#define dio0 9
#define ss 53

void setup() {
  Serial.begin(9600);
  while (!Serial);
 
  Serial.println("LoRa Receiver");
  LoRa.setPins(rst,dio0);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    delay(100);
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());

    delay(1000);
  }
}

IoTThinks, 2020-04-07:
DIO0 is mapping to gpio9 (in your code) or 8 (in your photo)?
The setPins should be `LoRa.setPins(ss, reset, dio0);`

alfantamamy, 2020-04-07:
> DIO0 is mapping to gpio9 (in your code) or 8 (in your photo)?
> The setPins should be `LoRa.setPins(ss, reset, dio0);`
> 
> 

Dio0 to pin 9.

alfantamamy, 2020-04-07:
> I understand that these modules are not working easily. First of all your hardware should be good one and then your code must be simplest one. Below is my tested code and works properly. When you [...]

in My Arduino can't read #include <LORA.h>
I just used #include <LoRa.h>
are not same?

IoTThinks, 2020-04-07:
The code is not from this library. Sure not working with this library.

Sx1278 is 3.3v.
If you ever put it into 5v, the chip sx1278 may be already damaged.

alfantamamy, 2020-04-07:
> The code is not from this library. Sure not working with this library.
> 
> Sx1278 is 3.3v.
> If you ever put it into 5v, the chip sx1278 may be already damaged.

Whats (name) the code on library, I'll download it. 

Yeah, I used 3.3v , because on lora board written using 3.3v.

Gwoud, 2021-12-23:
> Actually my problem was solved. I selected Lora normal mode. And now its working

i got the same issue. Please help me. How to select LoRa normal mode

EsKliwoN, 2022-01-24:
> Actually my problem was solved. I selected Lora normal mode. And now its working

how u solve it? and how to selected LoRa to Normal Mode

suryasarafe, 2022-09-01:
> > Actually my problem was solved. I selected Lora normal mode. And now its working
> 
> how u solve it? and how to selected LoRa to Normal Mode

Same issue here, I try with this code and this working now.

```
#include <SPI.h>
#include <LoRa.h>
 
#define ss 15
#define rst 16
#define dio0 2
 
void setup() {
  Serial.begin(115200);
  while (!Serial);
 
  Serial.println("LoRa Receiver Callback");
  LoRa.setPins(ss, rst, dio0);
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
 
void loop() {
  String str="";
  int packetSize = LoRa.parsePacket(),i=0;
  if (packetSize) {
    Serial.print("Received packet ");
    while (LoRa.available()) 
    {
      str=str+((char)LoRa.read()); 
    }
    Serial.println(str);
  }
}
```

I'm using nodeMCU and sx1278 (RA-02) this is the wiring
```
NodeMCU         SX1278
GND             GND
3.3v            3.3v
D0              RST
D8              NSS
D7              MOSI
D6              MISO
D5              SCK
```
hope this can help,

Axliyor, 2022-11-05:
> > The code is not from this library. Sure not working with this library.
> > Sx1278 is 3.3v.
> > If you ever put it into 5v, the chip sx1278 may be already damaged.
> 
> Whats (name) the code on library, I'll download it.
> 
> Yeah, I used 3.3v , because on lora board written using 3.3v.

Hi, I have the same problem. But the SPI pins run at 5V on the Arduino Mega. Through this, 5V is entering the ra-02!

fatans, 2024-12-11:
Hi, i have the same problem. But i can, you can also move the 3.3V pin to 5V pin