sandeepmistry/arduino-LoRa #27: False intrrupt trigger in lora arduino

varunhr, 2017-06-23:
Hi Im facing false interrupt triggers in lora library. Im using a Arduino DIP IC at 3.3V on bread board and DRF1276F Lora module. Have attached the interrupt to pin number 2 and after some 2 hours of running im getting a false trigger and after that the controller hangs. I even tried to enable the WDT but still it hangs in between with false triggers. could you please help me with this. I have attached my code below.

`#include <SPI.h>
#include <LoRa.h>
#include <avr/wdt.h>

#define NODE_ID 1
int counter = 0;
const byte numChars = 40;
char receive_buff[numChars];
char tempChars[numChars];        // temporary array for use by strtok() function
int from_address,to_address,firstArg,secondArg;
boolean loradata = false;
boolean newData = false;
boolean command_make_flag = false;
boolean pw_command_flag = false;
boolean ack_received_flag = false;
boolean read_command_flag = false;
boolean pr_command_flag = false;
char commandFromPC[10];
char pactet_type;
int pactet_number,soil_moisture,luminocity_val;
float temperature_val,humidity_val,pressure_val;
int port_state;

void setup() 
{
  pinMode(8,OUTPUT);
  pinMode(7,OUTPUT);
  digitalWrite(2, INPUT_PULLUP);    // Turn on internal Pull-Up Resistor
  Serial.begin(9600);
  while (!Serial);
  
  Serial.println("LoRa Receiver Callback");

  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  // register the receive callback
  LoRa.onReceive(onReceive);

  // put the radio into receive mode
  LoRa.receive();
  wdt_enable(WDTO_8S);
}
unsigned long now = millis();
unsigned long now1 = millis();
void loop() 
{
  // do nothing
  scan_flag();
  digitalWrite(7,HIGH);
  delay(50);
  digitalWrite(7,LOW);
  delay(50);
  wdt_reset();  

  while(millis() - now > 60000)
  {
        LoRa.beginPacket();
        LoRa.print("<1,0,0,22.76,45.3,911.76,433,250>");
//        LoRa.print(counter);
        LoRa.endPacket();
        LoRa.receive();
        Serial.println("<1,0,0,22.76,45.3,911.76,433,250>");
        now = millis();
  }

  while(millis() - now1 > 33000)
  {
    char message[20];    
    port_state = digitalRead(8);         
    sprintf(message,"<%d,%d,2,%d>",NODE_ID,from_address,port_state);
    LoRa.beginPacket();
    LoRa.print(message);
    LoRa.endPacket();
    LoRa.receive(); 
    Serial.println(message);
    now1 = millis();
  }
}

void onReceive(int packetSize) 
{
  int i = 0;
  // received a packet
  Serial.print("Received packet '");

  // read packet
  Serial.println(packetSize);

  for ( i = 0; i < packetSize; i++) 
  {
    receive_buff[i] = (char)LoRa.read();
  }
  i++;
  receive_buff[i] = '\0';
  // print RSSI of packet
  Serial.print("' with RSSI ");
  Serial.println(LoRa.packetRssi());
  loradata = true;
}

void scan_flag()
{
  
//  Serial.println("In Scan fun");
  if(loradata == true)
  {
//    Serial.println("lora data true");
    recvWithStartEndMarkers();
    loradata = false;
  }

  if(newData == true)
  {
   parseData();
   newData = false;
  }

  if(command_make_flag == true)
  {
    
    command_make();
//    command_make_flag = true;
  }

  if(pw_command_flag == true)
  {
    Serial.println("In PW command");
    digitalWrite(firstArg,secondArg); 
    pw_command_flag = false;
  }

//  if(read_command_flag == true)
//  {
//    Serial.println("In READ command");
////    readsoil();
//
//    sprintf(message,"<1,0,0,22.76,45.3,911.76,433,250>");
//    sendmsg();
//    read_command_flag = false;
//  }
//   if(pr_command_flag == true)
//  {
//    Serial.println("In PORT READ command");
////    readsoil();
//    port_state = digitalRead(firstArg);
//    sprintf(message,"<%d,%d,2,%d>",NODE_ID,from_address,port_state);
//    sendmsg();
//    pr_command_flag = false;
//  }
  
}

void parseData() 
{
//    Serial.print("In Parse Fun");
      // split the data into its parts
    char * strtokIndx; // this is used by strtok() as an index 
//   strcpy(tempChars, receive_buff);
//  strcpy(messageFromPC, strtokIndx); // copy it to messageFromPC    
//  integerFromPC = atoi(strtokIndx);     // convert this part to an integer

    strtokIndx = strtok(tempChars,",");      // get the first part - the string
    from_address = atoi(strtokIndx);     // convert this part to a float
  
    strtokIndx = strtok(NULL, ","); // this continues where the previous call left off
    to_address = atoi(strtokIndx);     // convert this part to a int
//    if(to_address == NODE_ID)
//    { 
      strtokIndx = strtok(NULL, " ");
      strcpy(commandFromPC, strtokIndx);     // convert this part to a float
  
      strtokIndx = strtok(NULL, " ");
      firstArg = atoi(strtokIndx);     // convert this part to a float
  
      strtokIndx = strtok(NULL, " ");
      secondArg = atoi(strtokIndx);

      Serial.print("From Addr is ");
      Serial.print(from_address);
      Serial.print("\n");
      Serial.print("To Addr is ");
      Serial.print(to_address);
      Serial.print("\n");
      Serial.print("Command is ");
      Serial.print(commandFromPC);
      Serial.print("\n");
      Serial.println("First arg is ");
      Serial.print(firstArg);
      Serial.print("\n");
      Serial.println("second arg is ");
      Serial.print(secondArg);
      Serial.print("\n");
   
      newData = false;
      command_make_flag = true;
//    }
    
}

void recvWithStartEndMarkers()
{
//    Serial.println("In receive fun");
    static boolean recvInProgress = false;
    static byte ndx = 0;
    char startMarker = '<';
    char endMarker = '>';
    char rc;
//    Serial.println("In Receive Fun");
    while (loradata == true)
    {
//      Serial.println("In receive while loop");
        rc = receive_buff[ndx];
//        Serial.println(rc);
        if (recvInProgress == true)
        {
//          Serial.println("receive progress");
            if (rc != endMarker) 
            {
                tempChars[ndx-1] = rc;
//                Serial.println(tempChars[ndx]);
                ndx++;
                if (ndx >= numChars)
                {
                    ndx = numChars - 1;
                }
            }
            else
            {
                tempChars[ndx-1] = '\0'; // terminate the string
                recvInProgress = false;
                ndx = 0;
                newData = true;
                loradata = false;
            }
        }

        else if (rc == startMarker) 
        {
            recvInProgress = true;
            ndx++;
        }
//        yield();
    }
    Serial.println("Received data is");
    Serial.println(receive_buff);
    Serial.println("Parsed data is");
    Serial.println(tempChars);   
//    Serial.println("Size of Parsed data is");
//    Serial.println(sizeof(tempChars));
//  Serial.println(sizeof(receivedChars));
  
}

void command_make()
{             
       if(strcmp(commandFromPC, "PW") == 0)
      {
        Serial.println("In PW command");
        digitalWrite(firstArg,secondArg); 
        pw_command_flag = false;
      }

      if(strcmp(commandFromPC, "READ") == 0)
      {
        Serial.println("Command is READ");
//        delay(1000);
        LoRa.beginPacket();
        LoRa.print("<1,0,0,22.76,45.3,911.76,433,250>");
        LoRa.print(counter);
        LoRa.endPacket();
        LoRa.receive();
//        read_command_flag = true; 
      }
//
//      if(strcmp(commandFromPC, "PR") == 0)
//      {
////        Serial.println("Command is PORT READ");
//        pr_command_flag = true; 
//      }
      
      command_make_flag = false;
}
`

sandeepmistry, 2017-07-30:
@varunhr please provide a more detailed info on how to reproduce the issue, and a more simpler sketch.

sandeepmistry, 2017-09-05:
Closing this for now due to lack of activity.

closed 2017-09-05 by sandeepmistry