sandeepmistry/arduino-LoRa #28: About the range and configuration

eron93br, 2017-06-30:
Hello Sandeep,

First of all I would like to congratulate you for your contributtion with this LoRa library!

Probably this post it's not a real issue, actually is just a couple of questions. 

I have some questions about this library/implementation;

1) Could you tell me about the range using the SX1286 (with Adafruit breakout board)? I did some tests and I got 700 meters at maximum. 

2) Have you perfomed any latence test with the examples codes?

Thank you very much

Cheers!

eron93br, 2017-06-30:
Sorry, I forgot about the configuration...

Could you tell me if this library is configured to the maximum range possible?  (with slow data rate)

sandeepmistry, 2017-07-31:
> Could you tell me about the range using the SX1286 (with Adafruit breakout board)? I did some tests and I got 700 meters at maximum.

700 meters seems good, the most i got so far is 200m. Please see the API docs for the default radio parameters used: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md

closed 2017-07-31 by sandeepmistry

RajatJain4061, 2018-02-01:
I am getting only 100 meters range (Line of Sight). Please let me know what is the best configuration to get upto 700 meters range.