sandeepmistry/arduino-LoRa #29: Problem to initialize LoRa with nodemcu.

Alexfante, 2017-06-30:
Hi!

When I start LoRa, (! LoRa.begin (868E6)) my nodemcu does not start, I press the reset and it restarts itself. If I remove the LoRa boot from the program, it stops reinitializing. What could it be?

I tested it with 2 arduinos and it works. I used the example scketchs from the LoRa library.
[image]
[image]

Alexfante, 2017-07-04:
I've updated the links.
[image]

I may even be being naive, but I think it should work.

#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa Receiver");
  LoRa.setPins(8, 2, 1);
  if (!LoRa.begin(868E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}

Alexfante, 2017-07-07:
It's working, apparently my arduino installation on windows 10 is in trouble. I made the same build in Fedora and it's working.

closed 2017-07-07 by Alexfante

fxsh1301, 2017-07-20:
@Alexfante 
Hi sir!
Do you have any project that using LoRa with Raspberry Pi ?