sandeepmistry/arduino-LoRa #33: how to get data from child lora devices to parent lora

tvjmohan, 2017-07-26:
I have tried the simple sender and receiver. its working fine. thank you for the detailed explanation.

i wold like to know how this communication is working with out mentioning any address to these devices. ?

and i wanted to know how the communication can be done from single to multiple lora devices ?

sandeepmistry, 2017-07-31:
> i wold like to know how this communication is working with out mentioning any address to these devices. ?
> and i wanted to know how the communication can be done from single to multiple lora devices ?

LoRa at the radio level is all broadcast, no addressing. You'll have to build you're own addressing scheme into the data payload.

closed 2017-07-31 by sandeepmistry

tvjmohan, 2017-08-02:
ok thank you. sandeep.