sandeepmistry/arduino-LoRa #34: Arduino Nano stops sending after 1 or 2 packets

JonSilver, 2017-07-29:
I've hooked up an RFM95W to an Arduino Uno, a Wemos D1 Mini and an Arduino Nano. This library works well with the Uno and Wemos. The simple Sender and Receiver examples work as expected. But on Nanos (I've tried several) it's fine receiving but when I run the sender it transmits the first packet, and displays the Serial debug message for the second packet, but fails to transmit it, and then just hangs there doing nothing. The on-board LED stops blinking and remains solid - is this something odd with the SPI clock and pin 13 LED on the Nano?

Simple answer - use anything but a Nano. But I'm hoping there's a better solution. :)

sandeepmistry, 2017-07-31:
@JonSilver maybe it's power supply related? 

Do you have another power source to try out with? I'm thinking the Nano can't supply enough current or something.

Sauro98, 2017-08-24:
I'm having the same issue too (even though I'm using the chinese version of the arduino nano and I'm using a a SX1276 LoRa chip).    
 
I have a nano set up as the receiver (and it also sends every 5 minutes) and it has been working for weeks straight.

Then I have other 2 nanos who have to send every three seconds and they will usually stop after 3/4 packets sent. 

I have to restart them many times and they they work for days before stopping again.

Do you think it could be power related too?

Sauro98, 2017-08-24:
More in the specific I found that it is always blocking in this loop inside `endPacket()`:

`while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);`

Sauro98, 2017-08-24:
My nano (with a sx1276 LoRa chip) at random times starts to loop endlessly at this line in `endPacket()`:

```
  // wait for TX done
  while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);
```

Sauro98, 2017-08-24:
If those can help here are the regdump just before the ine of code mentioned above. The right column ended with a succesful sending, the left one ended with an infinite loop

```
UNHEALTHY DUMP                                                     HEALTHY DUMP

0x0: 0x90                       REG_FIFO                                       V       0x0: 0x90
0x1: 0xF9                       REG_OP_MODE                              V       0x1: 0xF9
0x6: 0xF9                       REG_FRF_MSB                                V       0x6: 0xF9
0x7: 0xFC                       REG_FRF_MID                                V       0x7: 0xFC
0x8: 0xFC                       REG_FRF_LSB                                 V       0x8: 0xF9
0x9: 0x4F                       REG_PA_CONFIG                            X       0x9: 0x8F
0xC: 0x20                       REG_LNA                                       X       0xC: 0x23
0xD: 0x8                        REG_FIFO_ADDR_PTR                    X       0xD: 0xE
0xE: 0x2                        REG_FIFO_TX_BASE_ADDR             X       0xE: 0x0
0xF: 0xA                        REG_FIFO_RX_BASE_ADDR             X       0xF: 0x0
0x10: 0xFF                      REG_FIFO_RX_CURRENT_ADDR    X       0x10: 0x0
0x12: 0x15                      REG_IRQ_FLAGS                            X      0x12: 0x88
0x13: 0xB                       REG_RX_NB_BYTES                        X       0x13: 0xD
0x1A: 0x0                       REG_PKT_RSSI_VALUE                   X       0x1A: 0x86
0x1B: 0x0                       REG_PKT_SNR_VALUE                   X       0x1B: 0x2E
0x1D: 0x0                       REG_MODEM_CONFIG_1             X       0x1D: 0x72
0x1E: 0x0                       REG_MODEM_CONFIG_2             X        0x1E: 0x70
0x20: 0x0                       REG_PREAMBLE_MSB                  V        0x20: 0x0
0x21: 0x0                       REG_PREAMBLE_LSB                   X         0x21: 0x8
0x22: 0x0                       REG_PAYLOAD_LENGTH              X         0x22: 0xD
0x26: 0x3                       REG_MODEM_CONFIG_3            X         0x26: 0x4
0x2C: 0x55                      REG_RSSI_WIDEBAND               X         0x2C: 0x10
0x31: 0x40                      REG_DETECTION_OPTIMIZE      X         0x31: 0xC3
0x37: 0x0                       REG_DETECTION_THRESHOLD   X         0x37: 0xA
0x39: 0xF5                      REG_SYNC_WORD                     X         0x39: 0x12
0x40: 0x0                       REG_DIO_MAPPING_1                X         0x40: 0x0
0x42: 0x12                      REG_VERSION                            V        0x42: 0x12
```

kbaggott, 2017-08-25:
I'm also having this issue on a nano... any update other than "don't use a nano"?

Sauro98, 2017-08-28:
I've tried attaching the 3v3  pin in the LoRa module to another arduino (Uno) and it works longer, but then stops at some point.

pearteardial, 2017-08-29:
Hi Guys, 

I'm having the same issue. 

Did anyone have any luck solving it?

I tried reducing the SPI freq, but didn't have any success.

pearteardial, 2017-08-29:
Hi All. 

I think I've had some success changing 

`while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);`

to 

`// wait for TX done
  while((readRegister(REG_IRQ_FLAGS)) == 0);

  while(IRQ_TX_DONE_MASK == 0);
`

pearteardial, 2017-08-29:
Actually it was just missing a &.... I think. 

while((readRegister(REG_IRQ_FLAGS) && IRQ_TX_DONE_MASK) == 0);

Sauro98, 2017-08-29:
@pearteardial It is actually right to have only one `&` there because it needs to do a bitwise checking

pearteardial, 2017-08-29:
Oh sorry, 

still, I have found making that change has worked for me. 

Is my way valid from the point of view of outcome even if it's less efficient??

Sauro98, 2017-08-29:
@pearteardial they do a completely different thing.
`&` performs the check at bit level. For example:

```c
3 & 2 = 2
11 (3) &
10 (2) =
---------
10 (2)
```
`&&` performs the check at boolean logic level: For example:
```c
true && false = false;
true && true = true;
false && false = false;
```

pearteardial, 2017-08-29:
Ah I see, every bit is compared.

Sauro98, 2017-09-04:
# Update
I got myself a rasperry pi2,and wired the chip's 3v3 pin to the 3v3 pin on the raspberry. It works like a wonder, so it is indeed a power supply problem, I guess that the answer is just don't use a nano or, if you really have to, get yourself a decent power supply

sandeepmistry, 2017-09-05:
@Sauro98 awesome, thanks for trying that out. I'll make a note in the read me about this. The data sheet mentions `120 mA` is needed for `RFOP = +20 dBm, on PA_BOOST` (see `2.5.1. Power Consumption Table 6` section).

sandeepmistry, 2017-09-05:
Added note in https://github.com/sandeepmistry/arduino-LoRa/commit/7f1ca664499e2406ac03ceca9b22c4406a921cda.

closed 2017-09-05 by sandeepmistry

WoderfulIOT, 2018-10-21:
I am facing this issue and it seems related to antenna/RF perforamance. If antenna is changed the issue goes away. If antenna cable is removed the issue goes away. Does it mean this issue is only due to antenna mismatch and nothing else?

WoderfulIOT, 2018-10-21:
Please re-open this issue

WoderfulIOT, 2018-10-22:
Even when TX power is lowered, the device runs longer before hitting the problem

WoderfulIOT, 2018-10-23:
When the problem occurs REG_MODEM_CONFIG_1 becomes 0. During normal operation it is 0x82

pranav083, 2021-08-12:
@WoderfulIOT did you get your problem resolved regarding the power problem with the lora and nano.
Thanks for reply

matg41, 2022-12-31:
> @Sauro98 awesome, thanks for trying that out. I'll make a note in the read me about this. The data sheet mentions `120 mA` is needed for `RFOP = +20 dBm, on PA_BOOST` (see `2.5.1. Power Consumption Table 6` section).

It seems that this happens for esp32 too.

Portia-Lin, 2026-01-11:
> I am facing this issue and it seems related to antenna/RF performance. If antenna is changed the issue goes away. If antenna cable is removed the issue goes away. Does it mean this issue is only due to antenna mismatch and nothing else? Note:- I have a regulator powering LoRa (1A capacity) and a hefty 2200uF capacitor on the power supply so its not a power supply issue

Hello. I developed my board on the RA-02 module and found the same problem. The problem is in the antenna, or the position of the antenna, or the influence of my hand on my radio remote control. Sometimes the remote control works, sometimes it doesn't. But if I touch the antenna or the battery (hmmm?????) with my hand then the device definitely works. How can I solve the problem with the function `endpacket()`. Thank you.

JonSilver, 2026-01-11:
> Hello. I developed my board on the RA-02 module and found the same problem. The problem is in the antenna, or the position of the antenna, or the influence of my hand on my radio remote control. [...]

Antenna tuning is affected by everything in the antenna's environment, including casing materials, proximity of screws and other conductors (including hands), and the length and quality of any coax cable. You have to stabilise the antenna's environment in the same kind of state it'll experience in real world usage, then tune the antenna so it works.