sandeepmistry/arduino-LoRa #38: ESP32/ESP8266 Issues

r-devries, 2017-08-17:
Hi,  
I have been working with your LoRaDuplex sketch and I ran into some issues.
I was able to get it running with a Leonardo with a Dragino LoRa radio, I was also able to get it running with a M0 basic Feather and LoRa Feather. When I tried to get it working with an ESP32 I could never get the radio initialized, and when I tried to get it running on an ESP8266 it would just keep resetting. 
      Soft WDT reset
      ctx: cont
      sp: 3ffef9f0 end: 3ffefbd0 offset: 01b0
      >>>stack>>>
      3ffefba0: 3fffdad0 3ffeea78 3ffeeb7c 40201f65
      3ffefbb0: feefeffe 00000000 3ffeeb98 4020331c
      3ffefbc0: feefeffe feefeffe 3ffeebb0 40100114
      <<<stack<<<
I know the hardware works, but would there be an issue with the ESP32/ESP8266.  I wouldn't think so but this has me stumped.
Any thoughts?

Thanks,

FelixHr, 2017-08-17:
I am successfully using two LoRa sx1278 modules, these simple barebone ones from china, together with two Esp8266. 
I also ran into some problems with the watchdog timer and the Esp resetting. 
The issue was the while(true); loop if the Esp could not connect to the Lora module. The Esp gets stuck in the loop and the WDT timer resets the chip. 
Try adding a yield(); command in the loop to give the Esp time to perform other stuff in the background. 
Hope that helps!

r-devries, 2017-08-19:
Well I guess this one is user error.
I attached the CS to a wrong pin for both the ESP32 and the ESP8266.
The M0 I got correct and that's why it worked out of the gate.
Now I am able to send temperature and humidity readings.
Next step is to make one node a bridge for MQTT.
I will have to remember yield().
Thank you,

sandeepmistry, 2017-08-19:
Closing this as resolved for now. @r-devries @FelixHr pull requests welcome for the yield suggestion.

PR #61 (2017-12-18): stabilize ESP8266 watchdog
solve the watchdog timeout issue on ESP8266
fixes #38
closed 2017-08-19 by sandeepmistry