sandeepmistry/arduino-LoRa #39: Different bandwidths

kbaggott, 2017-08-26:
I'm using 433 as a frequency but can't seem to change the bandwidth to anything lower than the default, any ideas?

sandeepmistry, 2017-09-05:
@kbaggott could you please share a sketch to provide more details on what you are trying to do. Thanks.

kbaggott, 2017-09-05:
Just tried a few of the examples you had but with the `LoRa.setSignalBandwidth();` function set like below. If I use 125E3 / 250E3 it seems to work but anything lower than that and nothing is received.
I'm fully aware this may be nothing to do with your library but if you have any ideas they would be most appreciated.
(PS. I am setting the bandwidth at both ends in the same place just after `LoRa.begin()`

```
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setSignalBandwidth(31.25E3);
  //7.8E3, 10.4E3, 15.6E3, 20.8E3, 31.25E3, 41.7E3, 62.5E3, 125E3, and 250E3.
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
  }
}
```

sandeepmistry, 2017-09-11:
@kbaggott I don't have any 433 MHz devices here to test with, but could you make sure the registers look ok via `dumpRegisters` (https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.h#L63).

I don't see anything in the data sheet about 433 MHz not supporting lower bandwidths. You could try using other spreading factors as well.

sandeepmistry, 2017-09-28:
Closing this for now due to lack of feedback, please re-open with response if you are still interested.

closed 2017-09-28 by sandeepmistry