sandeepmistry/arduino-LoRa #42: LoRa Low Power

pearteardial, 2017-09-14:
Hey, I'm trying to set the sender to a low power mode either sleep or standby. The problem I'm having is that when i try to put the device in sleep or standby mode I see no reduction in power consumption on my multimeter... Anyone else experience this? 

`    //  temp = analogRead(thermistorpin);
    Serial.print("current temp: ");
    Serial.println(t);

    LoRa.beginPacket();
    LoRa.print(t);
    LoRa.endPacket();
    delay(2000);
    // send packet
    LoRa.beginPacket();
    LoRa.print(t);
    LoRa.endPacket();
    counter++;
  
    delay(5000);
      /* Don't forget to clear the flag. */
    f_wdt = 0;
    LoRa.sleep();
    Serial.println("Lorasleep");
      /* Re-enter sleep mode. */
    enterSleep();`

pearteardial, 2017-09-14:
It seems that when I read the register after going into sleep mode, the register returns an incorrect value. 
writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_SLEEP);
  int b = readRegister(REG_OP_MODE);
  return b;

it should return 0x80 but instead returns 0x82. Any suggestions on this?

dontsovcmc, 2017-12-20:
What RF module you use? Write full code here.

yrik, 2018-08-05:
@FabianInostroza Should it be modified in the code of the library? Struggling with the same issue. Any help would be great.

Tirititelu, 2018-11-04:
I found the file "wiring.c" and made the change (all pins to INPUT). Now I have a consumption of 1.15mA with a sketch that only sleeps the board with the function "LowPower.deepSleep ()" of the ArduinoLowPower library.

But if I add communications LoRa with the library "ArduinoLora.h" and its function "LoRa.begin ()" the consumption rises to 8.1mA even sleeping the radio with "LoRa.sleep ()", I can not get down to 1.15mA nor using the "LoRa.end ()" function after sending LoRa packet. This is strange. Do you know how to get the 1.15mA again after making a communication LoRa ?.

Thanks!

FabianInostroza, 2018-11-04:
@Tirititelu Are you measuring the LoRa module current only?, What LoRa module are you using?
Did you try what I wrote in my previous comment?

Tirititelu, 2018-11-05:
Sorry for the lack of information. I am using the Arduino MKR1300, of which I am measuring mAh.
I do not know if I am understanding very well what you wrote in your message. Do you mean LORA_IRQ_DUMB of the LoRa.begin () function?.

Thanks

mikaelsundin, 2019-02-23:
I got my RA-02 module into 0.6uA by go to LORA sleep and then FSK sleep.
With only regular sleep() i got 1.8mA, probably idle mode for SX1278.

```
void LoRaClass::shutdown(){
	if (_reset != -1) {
		pinMode(_reset, OUTPUT);

		// perform reset
		digitalWrite(_reset, LOW);
		delay(10);
		digitalWrite(_reset, HIGH);
		delay(10);
	}
	
	/* Swap to FSK mode to get module into sleep */
	writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_SLEEP);
	delay(5);
	writeRegister(REG_OP_MODE, MODE_SLEEP); 
}
```

FlaegelEmbed, 2019-10-10:
Yeah,
to stop SPI and reconfigure MISO,MOSI, SCK and CS is very important. We saved the last 700µA with with this config.

Heres sample code for STM(not suitable thread here, but maybe it help others ):
`    GPIO_InitTypeDef gpio_init_struct = {0};

    /////////////////////
    // LORA SPI
    gpio_init_struct.Pin  = GPIO_PIN_10 | GPIO_PIN_11 | GPIO_PIN_12;
    gpio_init_struct.Mode = GPIO_MODE_INPUT;
    gpio_init_struct.Pull = GPIO_PULLDOWN;
    HAL_GPIO_Init(GPIOC, &gpio_init_struct);

    /////////////////////
    // LORA SPI CS
    gpio_init_struct.Pin  = GPIO_PIN_2;
    gpio_init_struct.Mode = GPIO_MODE_INPUT;
    gpio_init_struct.Pull = GPIO_PULLUP;
    HAL_GPIO_Init(GPIOD, &gpio_init_struct);

`

Cicicok, 2020-11-02:
I need to resurrect this question because I'm bit confused about power consumption of .end() and .sleep() functions. **NOTE** I'm only measuring power consumption of the LoRa chip (in my case RFM95w), which is connected to esp32.

Here are mine measurements:
.sleep() = 1.6 uA
.end() = 90 uA
esp32 deepsleep = 1.95 mA

What I don't understand is why closing SPI connection (which is called from .end()) raises power consumption to 90uA. And absolutely baffled that after switching esp32 into deepsleep the radio module jumps to 1.8mA (as if deepsleep somehow resets the radio module).

IoTThinks, 2020-11-02:
1.8mA is idle current of LoRa.
When esp32 goes to deepsleep, RST seems to trigger to restart the LoRa. Then LoRa goes to idle mode => 1.8mA.

You can set RST of LoRa to input mode before ESP32 goes to deepsleep.

Cicicok, 2020-11-02:
OK, setting INPUT solved the main, deep sleep, issue, but why closing SPI connection changes power consumption as well?

IoTThinks, 2020-11-02:
Set the SS to INPUT, too.
Floating SS may set SPI to active.

Cicicok, 2020-11-02:
It wasn't the SS pin, but it gave me an idea.

When I looked at code of SPI.end() in esp32 repo it showed, that it detaches those 3 SPI pins (MOSI, MISO, CLK) and sets them as INPUT. This causes the observed increase in power consumption (from 1.6 uA to 90 uA) and can be easily solved by setting those 3 pins as INPUT_PULLUP.

But after switching into deepsleep the power consumption goes back into 90 uA. This has to be solved be configuring RTC GPIOs of esp32. There are two ways how to do it:
1. hold configuration of gpio (for example gpio12) in deep sleep using rtc_gpio_hold_en(GPIO_NUM_12)
2. enable pullup (of spi pins) in deepsleep, this has to be done using 2 functions rtc_gpio_pullup_en(GPIO_NUM_12) & rtc_gpio_pulldown_dis(GPIO_NUM_12);

**Note** this applies only to esp32 and not all gpio can be configured for RTC.

IoTThinks, 2020-11-03:
I also use RTC to keep state. But keeping state for RTC seems to take some power too.
At the end, I use and IC to turn off the power source of the LoRa chip completely.

IoTThinks, 2020-11-03:
> It wasn't the SS pin, but it gave me an idea.
>
> When I looked at code of SPI.end() in esp32 repo it showed, that it detaches those 3 SPI pins (MOSI, MISO, CLK) and sets them as INPUT. This causes the [...]

What is your GPIO12 linking to? SS/MISO/MOSI/SCK?

Cicicok, 2020-11-03:
GPIO12 is MISO on HSPI, because no GPIO of VSPI cannot be controlled by RTC. As I said I only measured consumption of RFM95W, not the board. But as alternative I was thinking to control power of LoRa chip using MOSFET.

closed 2021-06-11 by IoTThinks

chathura-de-silva, 2024-10-04:
Do Anyone here can give me an idea what does sleep() and idle() do here. What is the mode I should use if I am to put the ESP in to deep sleep mode and wake it up via DIO0 signal from the RX 1278(Lora Module)?