sandeepmistry/arduino-LoRa #43: Not a problem just a question (ATtiny85)

Sn0ff, 2017-09-14:
Hi, and thanks for the LoRa library. it can works with ATtiny ?
Thanks.

sandeepmistry, 2017-09-28:
@Sn0ff it should work, I haven't tested myself though. Please try it out and let us know.

closed 2017-09-28 by sandeepmistry

iotlearner0level, 2018-04-05:
the above output is for the lorasender arduino sketch:

#include <SPI.h>
#include <LoRa.h>
#define HARDWARE_SPI 1 
int counter = 0;

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Sender");
      SPI.begin();                   // start hardware SPI (the library sets up the clock and data pins)

  // override the default CS, reset, and IRQ pins (optional)
   LoRa.setPins(4, 0, 3); // set CS, reset, IRQ pin

/*  if (!LoRa.begin(866E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
*/
 if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

3Dtj, 2018-04-30:
iotlearner0level
Did you get the above sketch to work with the ATtiny?

3Dtj

iotlearner0level, 2018-05-01:
3Dtj To be honest, i gave up. instead i went for stm32/bluepill board which is cheap yet quite powerful and has lots of gpio pins and also spi/i2c/usb/can etc. I used the popular LMIC library (for LoRaWAN, TTN and not just sender/receiver). there were some problems initially, but they were sorted out after some help from the forums. i would recommend this to begin with: https://www.thethingsnetwork.org/labs/story/a-cheap-stm32-arduino-node

3Dtj, 2018-05-01:
Thanks!  I'm reviewing the links that you posted for Clemens Riederer and I have to believe that if he got his project working then I should be able to get Sandeep's code to run off of the ATTiny.  I'll check the stm32 suggestion too.

iotlearner0level, 2018-05-01:
3Dtj

Please keep us informed. however, i think the real power of lora is the loraWan "network" That makes it "internet" of things. for point-to-point connections, there are many cheaper alternatives. ATtiny85 has limited number of pins. so even if you get it running, i'm not sure if that will be useful for sending any practical data, if cheap is your goal. Some people suggested attiny84, apparentlly, not that popular in my country i can get bluepill cheaper. 
 if you want something which works on AT commands directly over serial, i think they are somewhat costlier. The things network sometimes behaves wierd, but i think if data reaches the network, then there are many "integrations" so it is easier to work. This is one advantage over point to point.

3Dtj, 2018-05-02:
Agree on the leveraging the network which is my goal.  I just wanted to
ensure I could accomplish the supposedly simple step of utilizing the
ATtiny85 in place of the Ardunio UNO before tackling the LoRaWAN
challenge.  :)

iotlearner0level, 2018-05-02:
ok, point taken. i applied similar approach too ;-) With all that said, i would say that i'm still interested!

3Dtj, 2018-05-06:
I'm still stuck even though I have everything wired up.  I'm checking
connections, components, etc.

Since we have to compile the sketch to the ATtiny, do you think we would
need the actual SPI library code included since the ATtiny is unable to
reach back to the library folder as would be the case if using an Arduino
connected to a PC?   That's what I'm pondering as the problem.  Thoughts?

iotlearner0level, 2018-05-17:
thanks for sharing. i would try that in a month or so when i have some more time....i think it is a great news. hope to make something useful. Attiny85 is quite cheap and really tiny ;-)

Abd-Bin, 2020-10-24:
Hi,

I am trying to use the ATTINY85 with my sx1276 transceiver. I have already tried  the simple receiver and sender code by Sandeep and it worked fine using two Arduino boards at both ends. 
Now at the transmitting side I am using the ATTiny85-SX1276 setup, but the packet is not being received at the Nano-SX1276 setup. I have also tried swapping the MOSI and MISO pins as mentioned above but still no luck.
Any Help?

Thanks!