sandeepmistry/arduino-LoRa #45: Block/Freeze when sending packet

Franciscopcosta, 2017-09-19:
Hi, my sketch it stop working when it reaches this lines:
```
  LoRa.beginPacket();                   // start packet
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
```
Do anybody had this issue before?

The example works OK, but when i adapt to my code, it blocks in this lines, it doesn't even send the packet.

sjtupkar, 2017-09-23:
My sketch running on Arduino hangs when it reaches the line LoRa.endPacket();
The issue is intermittent. The stall occurs after transmitting some packets. Sometimes immediately after couple of packets, sometimes after 10-15 packets.
By putting debug prints, I found that every stall occurs when the LoRa.endPacket is called.

I already shifted the 3.3V supply from Arduino to Raspberry Pi to ensure required current to LoRa module.

My sketch is farily simple.
int LoraData;
// some small code that sets value LoraData based on GPIO values.
LoRa.beginPacket();
LoRa.print(LoraData);
LoRa.endPacket();

sandeepmistry, 2017-09-28:
@Franciscopcosta please provide a minimal sketch to reproduce your issue, given the examples works it's most likely something in your sketch.

Franciscopcosta, 2017-10-14:
`#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
#include <SPI.h>
#include <LoRa.h>

SoftwareSerial mySerial(2, 3); // RX, TX

//String MQTT_HOST = "m10.cloudmqtt.com";
//String MQTT_PORT = "12385";

String MQTT_HOST = "178.62.119.199";
String MQTT_PORT = "1883";

void setLoraTX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, HIGH);  //TxEN
  digitalWrite(7, LOW); //RxEN
}

void setLoraRX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, HIGH); //RxEN
}

void GSM_MQTT::AutoConnect(void) {
  //(ClientIdentifier,UserNameFlag,PasswordFlag,UserName,Password,CleanSession,WillFlag,WillQoS,WillRetain,WillTopic,WillTopic)
  connect("GSM800L01", 1, 1, "admin", "admin", 1, 0, 0, 0, "", "");
  //connect("GSM800L02", 1, 1, "ihyoofdv", "__UzGYSOaNN8", 1, 0, 0, 0, "", "");
}

void GSM_MQTT::OnConnect(void) {
  //subscribe(0, _generateMessageID(), "SampleTopic", 1);
  //subscribe(0, _generateMessageID(), "boat/001/message_operator", 1); //(DUP,Message ID,SubTopic,SubQoS)
  subscribe(0, _generateMessageID(), "boat/002/message_operator", 1);
  //publish(0, 0, 0, _generateMessageID(), "SampleTopic", "Hello"); //(DUP,QoS RETAIN,Message ID,Topic,Message)
}

void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength) {
  digitalWrite(5, HIGH);
  delay(30);
  digitalWrite(5, LOW);
  setLoraTX();
  String payload2Send = "002";
  payload2Send += (String)Message;
  sendMessage(payload2Send);
  setLoraRX();
  //LoRa.receive();

  //  mySerial.println("LoRa sent!");
  //
  //  mySerial.println(TopicLength);
  //  mySerial.println(Topic);
  //  mySerial.println(MessageLength);
  //  mySerial.println(Message);
  bip(1,100);
}

GSM_MQTT MQTT(100); //20 is the keepalive duration in seconds

void setup() {
  pinMode(8, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(5, OUTPUT);
  setLoraRX();
  delay(100);
[... 156 lines cut ...]
      payload += ", \"SNR\": ";
      payload += SNR;
      payload += " }";
      payload.toCharArray(dataRcvvdd, (payload.length() + 1));
      delay(10);

      String topic = "boat/";
      topic += String(deviceId);
      topic += "/location";

      char topicName[20];
      topic.toCharArray(topicName, (topic.length() + 1));

      //    Serial.println(" topicname: " + topic);
      delay(10);
      MQTT.publish(0, 0, 0, "lora1234123", topicName, dataRcvvdd);
      //MQTT.publish(0, 0, 0, "lora2", "boat/001/location", "dummy");
      //MQTT.publish("boat/zeMinhoca/location", data);
      delay(10);

    }

  }
  }
*/`

Franciscopcosta, 2017-10-14:
#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
#include <SPI.h>
#include <LoRa.h>

SoftwareSerial mySerial(2, 3); // RX, TX

//String MQTT_HOST = "m10.cloudmqtt.com";
//String MQTT_PORT = "12385";

String MQTT_HOST = "178.62.119.199";
String MQTT_PORT = "1883";

void setLoraTX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, HIGH);  //TxEN
  digitalWrite(7, LOW); //RxEN
}

void setLoraRX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, HIGH); //RxEN
}

void GSM_MQTT::AutoConnect(void) {
  //(ClientIdentifier,UserNameFlag,PasswordFlag,UserName,Password,CleanSession,WillFlag,WillQoS,WillRetain,WillTopic,WillTopic)
  connect("GSM800L01", 1, 1, "admin", "admin", 1, 0, 0, 0, "", "");
  //connect("GSM800L02", 1, 1, "ihyoofdv", "__UzGYSOaNN8", 1, 0, 0, 0, "", "");
}

void GSM_MQTT::OnConnect(void) {
  //subscribe(0, _generateMessageID(), "SampleTopic", 1);
  //subscribe(0, _generateMessageID(), "boat/001/message_operator", 1); //(DUP,Message ID,SubTopic,SubQoS)
  subscribe(0, _generateMessageID(), "boat/002/message_operator", 1);
  //publish(0, 0, 0, _generateMessageID(), "SampleTopic", "Hello"); //(DUP,QoS RETAIN,Message ID,Topic,Message)
}

void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength) {
  digitalWrite(5, HIGH);
  delay(30);
  digitalWrite(5, LOW);
  setLoraTX();
  String payload2Send = "002";
  payload2Send += (String)Message;
  sendMessage(payload2Send);
  setLoraRX();
  //LoRa.receive();

  //  mySerial.println("LoRa sent!");
  //
  //  mySerial.println(TopicLength);
  //  mySerial.println(Topic);
  //  mySerial.println(MessageLength);
  //  mySerial.println(Message);
  bip(1,100);
}

GSM_MQTT MQTT(100); //20 is the keepalive duration in seconds

void setup() {
  pinMode(8, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(5, OUTPUT);
  setLoraRX();
  delay(100);
[... 156 lines cut ...]
      payload += ", \"SNR\": ";
      payload += SNR;
      payload += " }";
      payload.toCharArray(dataRcvvdd, (payload.length() + 1));
      delay(10);

      String topic = "boat/";
      topic += String(deviceId);
      topic += "/location";

      char topicName[20];
      topic.toCharArray(topicName, (topic.length() + 1));

      //    Serial.println(" topicname: " + topic);
      delay(10);
      MQTT.publish(0, 0, 0, "lora1234123", topicName, dataRcvvdd);
      //MQTT.publish(0, 0, 0, "lora2", "boat/001/location", "dummy");
      //MQTT.publish("boat/zeMinhoca/location", data);
      delay(10);

    }

  }
  }
*/

Franciscopcosta, 2017-10-14:
#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
#include <SPI.h>
#include <LoRa.h>

SoftwareSerial mySerial(2, 3); // RX, TX

//String MQTT_HOST = "m10.cloudmqtt.com";
//String MQTT_PORT = "12385";

String MQTT_HOST = "178.62.119.199";
String MQTT_PORT = "1883";

void setLoraTX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, HIGH);  //TxEN
  digitalWrite(7, LOW); //RxEN
}

void setLoraRX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, HIGH); //RxEN
}

void GSM_MQTT::AutoConnect(void) {
  //(ClientIdentifier,UserNameFlag,PasswordFlag,UserName,Password,CleanSession,WillFlag,WillQoS,WillRetain,WillTopic,WillTopic)
  connect("GSM800L01", 1, 1, "admin", "admin", 1, 0, 0, 0, "", "");
  //connect("GSM800L02", 1, 1, "ihyoofdv", "__UzGYSOaNN8", 1, 0, 0, 0, "", "");
}

void GSM_MQTT::OnConnect(void) {
  //subscribe(0, _generateMessageID(), "SampleTopic", 1);
  //subscribe(0, _generateMessageID(), "boat/001/message_operator", 1); //(DUP,Message ID,SubTopic,SubQoS)
  subscribe(0, _generateMessageID(), "boat/002/message_operator", 1);
  //publish(0, 0, 0, _generateMessageID(), "SampleTopic", "Hello"); //(DUP,QoS RETAIN,Message ID,Topic,Message)
}

void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength) {
  digitalWrite(5, HIGH);
  delay(30);
  digitalWrite(5, LOW);
  setLoraTX();
  String payload2Send = "002";
  payload2Send += (String)Message;
  sendMessage(payload2Send);
  setLoraRX();
  //LoRa.receive();

  //  mySerial.println("LoRa sent!");
  //
  //  mySerial.println(TopicLength);
  //  mySerial.println(Topic);
  //  mySerial.println(MessageLength);
  //  mySerial.println(Message);
  bip(1,100);
}

GSM_MQTT MQTT(100); //20 is the keepalive duration in seconds

void setup() {
  pinMode(8, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(5, OUTPUT);
  setLoraRX();
  delay(100);
  MQTT.begin();
  LoRa.setPins(10, 9, 30);
  LoRa.setSpreadingFactor(12);
  LoRa.enableCrc();
  //LoRa.setTxPower(30);
  if (!LoRa.begin(868E6)) {
    mySerial.println("Starting LoRa failed!");
    while (1);
  }
  LoRa.receive();
  LoRa.onReceive(onReceive);
}

void sendMessage(String outgoing) {
  bip(2,200);
  delay(550);
  LoRa.beginPacket();                   // start packet
  bip(3,300);
  LoRa.print(outgoing);                 // add payload
  bip(4,400);
  LoRa.endPacket();                     // finish packet and send it
  delay(550);
  //  delay(200);
  //  LoRa.beginPacket();                   // start packet
  //  LoRa.print(outgoing);                 // add payload
  //  LoRa.endPacket();                     // finish packet and send it
  //  delay(200);
  //  LoRa.beginPacket();                   // start packet
  //  LoRa.print(outgoing);                 // add payload
  //  LoRa.endPacket();                     // finish packet and send it

  bip(10,50);
}

//unsigned long previousMilliss = 0;
//const long interval = 2000;

void bip(int bips, int duration) {
  for (int i = 0; i < bips; i++) {
    digitalWrite(5, HIGH);
    delay(duration);
    digitalWrite(5, LOW);
    delay(duration);
  }
}

void loop() {
  //unsigned long currentMillis = millis();
  //onReceive(LoRa.parsePacket());
  if (MQTT.available())
  {
    onReceive(LoRa.parsePacket());
  }
  MQTT.processing();
  delay(1);
  yield();
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;
  if (packetSize) {
    //Serial.print("Received packet '");
    String RSSIvalue = String(LoRa.packetRssi());
    String SNR = String(LoRa.packetSnr());
    String rcvdPayload = "";
    while (LoRa.available()) {
      rcvdPayload += (char)LoRa.read();
    }
    rcvdPayload += "/";
    rcvdPayload += RSSIvalue;
    rcvdPayload += "/";
    rcvdPayload += SNR;

    char payload[50];
    rcvdPayload.toCharArray(payload, (rcvdPayload.length() + 1));

    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '1' && payload[4] != 'M' && payload[4] != 'P') MQTT.publish(0, 0, 0, "lora1", "boat/001/location", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '1' && payload[4] == 'P') MQTT.publish(0, 0, 0, "lora1", "boat/001/distress", payload);

    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2') MQTT.publish(0, 0, 0, "lora1", "boat/002/location", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'M') MQTT.publish(0, 0, 0, "lora1", "boat/002/message_boat", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'P' && payload[5] != '0') MQTT.publish(0, 0, 0, "lora1", "boat/002/distress", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'P' && payload[5] == '0' && payload[6] == '0') MQTT.publish(0, 0, 0, "lora1", "boat/002/distress_end", "off");
  }
}

Franciscopcosta, 2017-10-14:
```
#include "GSM_MQTT.h"
#include <SoftwareSerial.h>
#include <SPI.h>
#include <LoRa.h>

SoftwareSerial mySerial(2, 3); // RX, TX

//String MQTT_HOST = "m10.cloudmqtt.com";
//String MQTT_PORT = "12385";

String MQTT_HOST = "178.62.119.199";
String MQTT_PORT = "1883";

void setLoraTX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, HIGH);  //TxEN
  digitalWrite(7, LOW); //RxEN
}

void setLoraRX() {
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, LOW); //RxEN
  delay(100);
  digitalWrite(8, LOW);  //TxEN
  digitalWrite(7, HIGH); //RxEN
}

void GSM_MQTT::AutoConnect(void) {
  //(ClientIdentifier,UserNameFlag,PasswordFlag,UserName,Password,CleanSession,WillFlag,WillQoS,WillRetain,WillTopic,WillTopic)
  connect("GSM800L01", 1, 1, "admin", "admin", 1, 0, 0, 0, "", "");
  //connect("GSM800L02", 1, 1, "ihyoofdv", "__UzGYSOaNN8", 1, 0, 0, 0, "", "");
}

void GSM_MQTT::OnConnect(void) {
  //subscribe(0, _generateMessageID(), "SampleTopic", 1);
  //subscribe(0, _generateMessageID(), "boat/001/message_operator", 1); //(DUP,Message ID,SubTopic,SubQoS)
  subscribe(0, _generateMessageID(), "boat/002/message_operator", 1);
  //publish(0, 0, 0, _generateMessageID(), "SampleTopic", "Hello"); //(DUP,QoS RETAIN,Message ID,Topic,Message)
}

void GSM_MQTT::OnMessage(char *Topic, int TopicLength, char *Message, int MessageLength) {
  digitalWrite(5, HIGH);
  delay(30);
  digitalWrite(5, LOW);
  setLoraTX();
  String payload2Send = "002";
  payload2Send += (String)Message;
  sendMessage(payload2Send);
  setLoraRX();
  //LoRa.receive();

  //  mySerial.println("LoRa sent!");
  //
  //  mySerial.println(TopicLength);
  //  mySerial.println(Topic);
  //  mySerial.println(MessageLength);
  //  mySerial.println(Message);
  bip(1,100);
}

GSM_MQTT MQTT(100); //20 is the keepalive duration in seconds

void setup() {
  pinMode(8, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(5, OUTPUT);
  setLoraRX();
  delay(100);
  MQTT.begin();
  LoRa.setPins(10, 9, 30);
  LoRa.setSpreadingFactor(12);
  LoRa.enableCrc();
  //LoRa.setTxPower(30);
  if (!LoRa.begin(868E6)) {
    mySerial.println("Starting LoRa failed!");
    while (1);
  }
  LoRa.receive();
  LoRa.onReceive(onReceive);
}

void sendMessage(String outgoing) {
  bip(2,200);
  delay(550);
  LoRa.beginPacket();                   // start packet
  bip(3,300);
  LoRa.print(outgoing);                 // add payload
  bip(4,400);
  LoRa.endPacket();                     // finish packet and send it
  delay(550);
  //  delay(200);
  //  LoRa.beginPacket();                   // start packet
  //  LoRa.print(outgoing);                 // add payload
  //  LoRa.endPacket();                     // finish packet and send it
  //  delay(200);
  //  LoRa.beginPacket();                   // start packet
  //  LoRa.print(outgoing);                 // add payload
  //  LoRa.endPacket();                     // finish packet and send it

  bip(10,50);
}

//unsigned long previousMilliss = 0;
//const long interval = 2000;

void bip(int bips, int duration) {
  for (int i = 0; i < bips; i++) {
    digitalWrite(5, HIGH);
    delay(duration);
    digitalWrite(5, LOW);
    delay(duration);
  }
}

void loop() {
  //unsigned long currentMillis = millis();
  //onReceive(LoRa.parsePacket());
  if (MQTT.available())
  {
    onReceive(LoRa.parsePacket());
  }
  MQTT.processing();
  delay(1);
  yield();
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;
  if (packetSize) {
    //Serial.print("Received packet '");
    String RSSIvalue = String(LoRa.packetRssi());
    String SNR = String(LoRa.packetSnr());
    String rcvdPayload = "";
    while (LoRa.available()) {
      rcvdPayload += (char)LoRa.read();
    }
    rcvdPayload += "/";
    rcvdPayload += RSSIvalue;
    rcvdPayload += "/";
    rcvdPayload += SNR;

    char payload[50];
    rcvdPayload.toCharArray(payload, (rcvdPayload.length() + 1));

    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '1' && payload[4] != 'M' && payload[4] != 'P') MQTT.publish(0, 0, 0, "lora1", "boat/001/location", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '1' && payload[4] == 'P') MQTT.publish(0, 0, 0, "lora1", "boat/001/distress", payload);

    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2') MQTT.publish(0, 0, 0, "lora1", "boat/002/location", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'M') MQTT.publish(0, 0, 0, "lora1", "boat/002/message_boat", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'P' && payload[5] != '0') MQTT.publish(0, 0, 0, "lora1", "boat/002/distress", payload);
    if (payload[0] == '0' && payload[1] == '0' && payload[2] == '2' && payload[4] == 'P' && payload[5] == '0' && payload[6] == '0') MQTT.publish(0, 0, 0, "lora1", "boat/002/distress_end", "off");
  }
}
```

sandeepmistry, 2017-11-13:
@Franciscopcosta can you please further simply the sketch you provided to a minimal case to reproduce the issue.

sandeepmistry, 2017-12-18:
Closing this for now due to lack of activity. Please provide requested info. if you are still interested.

closed 2017-12-18 by sandeepmistry

bkomac, 2018-03-03:
For future reference ... Same issue with me. Must be a power problem. 
Try powering down tx with LoRa.setTxPower(2) to minimum right after LoRa.begin.

tp035741, 2018-07-21:
Im facing the same problem after many tries and error i found out that LoRa.endPacket(); stop my loop so only one packet is sent. i tried it with other codes of mine it worked normally and i received all the data in the gateway. can i get help for this one please :)

xtrinch, 2019-12-06:
Having the same issue, most often it hangs after the first packet being sent, but sometimes it manages to send a couple (up to maybe 20) and then freezes

PeterPodhraczky, 2021-02-20:
I had the same issue with TTGO Lora32 Board. I loaded the "Transmitter" example app from https://github.com/YogoGit/TTGO-LORA32-V1.0.

It can send the first message, then will stuck at sending the second message. After removing all other lines of code for the LCD display and OLED, it works for me.

I suspect the issue is due to both LoRa and the Display is utilizing the same SPI channel/pins