sandeepmistry/arduino-LoRa #47: Poor RSSI

johnhmacleod, 2017-09-28:
Hi - thanks for this library.  I have been using other libraries to drive my inAIR9 modules (SX1276) and it was working just fine but is overly complex.  I can reach several km no problem.  However, when I use your library, I can't even reach 10m!  Reported RSSI when the tx & rx are side-by-side is -95!  And the messages die less than 10m apart.  I'm using your examples with some code to ensure parameters are set the same as I was using previously.

 ```
   if (!LoRa.begin(92252E4)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  else
    Serial.println("Started LoRa OK");
  LoRa.setSpreadingFactor(12);           // ranges from 6-12,default 7 see API docs
  LoRa.setTxPower(17);
  LoRa.setSignalBandwidth(125E3);
  LoRa.setCodingRate4(5);
```

Any ideas?

johnhmacleod, 2017-09-28:
...and the issue seems to be on the tx side.  If I replace the tx code with my old code using the previous library but keeping your rx code, RSSI goes up to -24 side-by-side & -65 at 10m & reception is fine too (-85) at 20m.

sandeepmistry, 2017-09-28:
@johnhmacleod could you please try adding a 2nd parameter to `setTxPower`, see https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power

johnhmacleod, 2017-10-01:
Thanks Sandeep - that makes a big difference!  I guess the inAIR9 is one board where RF is connected to RFO rather than PABoost.  I'm still getting slightly poorer RSSI numbers reported with this library but that may be due to some differences in the way the different chips report it.  I've not done any decent range tests yet to compare them.

sandeepmistry, 2017-10-02:
@johnhmacleod thanks for trying!

I'll close this for now, but if you have time maybe you can dump the registers in this library (https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDumpRegisters/LoRaDumpRegisters.ino) and compare to the other lib you use, there might be something we can tweak to make them comparable :D

closed 2017-10-02 by sandeepmistry

johnhmacleod, 2017-10-02:
> blob/master/examples/LoRaDumpRegisters/LoRaDumpRegisters.ino) and compare

Sure - it was already on my to do list!

John

cassiolucas, 2018-02-22:
Hello @johnhmacleod and @sandeepmistry  I am having the same issue, very poor signal strenght. I have added the LoRa.setTxPower(20) on the main loop() but I dont know if this is correct.

Can you please share the code where you got a higher signal power?

I am using NiceRF1276 V2.0

Thank you!

cassiolucas, 2018-02-22:
I have also tried the LoRa.setTxPower(17); and the best RSSI value I got is -66 and it less than 1 meter distance =/.

johnhmacleod, 2018-02-24:
Hi

You need to ensure the chip is using the right output pin.  See here  -

https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md

I used

LoRa.setTxPower(17, PA_OUTPUT_RFO_PIN);

HamedJafarzadeh, 2020-01-20:
Hi , I'm experiencing the same problem. But I did these two steps and my receiving RSSI improved so much . I'm using SX1276RF1IAS and my desired frequency is 868MHZ. Initially I was getting more or less ~90. Then I change this function in begin method to LoRa.setTxPower(17, PA_OUTPUT_RFO_PIN);  and I got to ~75.
Then I continued investigating the problem and I found out that connecting FEM_CTX to RXTX pin on my module, increased RSSI significantly. Now I'm getting around ~45.
I hope this could help someone out.

IoTThinks, 2020-01-21:
So good the RSSI at - 35.
I usually get RSSI at -50 to -70 with AI-Thinker only.

HamedJafarzadeh, 2020-01-30:
Just to add more details to the issue 
Today I decided to use 169MHz frequency and again I encountered same issue . Poor RSSI, But after spending two days, I managed to get around -18dBm at both 868MHz and 169MHz. but here are my findings about  SX1276RF1IAS modules .
FEM_CTX and FEM_CPS control the RF switches for antennas and they were the solution to my poor RSSI problem. 
if you are transmitting something at high frequency ( more than 500 MHz )
   FEM_CTX = 1
   
else if you are transmitting at low frequency
   FEM_CPS = 0

if you are receiving at high frequency
   FEM_CTX = 0
else if you are receiving at low frequency
   FEM_CPS = 1

All my finding happens when I look at this source code : 
[https://github.com/mixly/SMeshStudio/blob/2c7eb9baf8504256b8e7ecf154b295cbe672e511/mcore/plib/mbed-SX127x/sx127x.cpp]

I want to blame Semtech at this point, since they didn't provide any technical description for SX1276RF1IAS modules and I waste several days finding out these .

IoTThinks, 2020-01-31:
@HamedJafarzadeh The code says sx127x. So it may be applicable to SX1278?
I checked the datasheet for SX1276/77/78/79 - 137 MHz to 1020 MHz and don't know any registers for FEM_CTX and FEM_CPS.

Do you have any idea?
Thanks a lot.

HamedJafarzadeh, 2020-01-31:
They are pins on SX1276RF1IAS module for controlling the RF Switches.
You can see them by looking at following picture. 
Depending on your module, these pins can be removed by designer. by in my case, since my module has two antennas, I need to switch between antennas using them.
[image]

HamedJafarzadeh, 2020-01-31:
@IoTThinks regarding the code, I think most probably they have same registers since their datasheet are same and their registers are same. Use it with caution. I'm using Sandeep library, but I had to make some changes to work with my modules.

HamedJafarzadeh, 2020-02-17:
Just to add, we bought new SX1276RF1IAS module to make sure we didn't burn the modules accidentally . We got the same results. the best RSSI I achieved in about 1 meter distance between modules was -20dbm . I will use a spectrum analyzer to investigate the problem. In mean while we are using PyCom modules ( specifically LoPy4 ) for our test. It works flawlessly, at same range with 14dbm transmit power, it receive signal at ~1dbm !

IoTThinks, 2020-02-18:
> In mean while we are using PyCom modules ( specifically LoPy4 ) for our test. It works flawlessly, at same range with 14dbm transmit power, it receive signal at ~1dbm !
~ 1dBm means the signal is boosted more than it losses.
If the loss is around -4dBm, and the antenna is boosting +5dBm.
-4dBm sounds to good for me. Very keen to know more.

Are you using the same LoRa library for SX1276RF1IAS and PyCom modules?
Did you use an external measure tool to measure the RSSI or internal function of LoRa library to get RSSI of received signal?

ned-kelly, 2020-09-27:
> Thanks Sandeep - that makes a big difference! I guess the inAIR9 is one board where RF is connected to RFO rather than PABoost. I'm still getting slightly poorer RSSI numbers reported with this [...]

Hi @johnhmacleod - What did you actually set here - Can you paste a snippet of your code by any chance?

michalmutka, 2021-04-01:
> @IoTThinks regarding the code, I think most probably they have same registers since their datasheet are same and their registers are same. Use it with caution. I'm using Sandeep library, but I had to make some changes to work with my modules.

 I found you were commenting this post about SX1276RF1IAS. Would you be willing to answer some questions
regarding this module? I am working on my bachelor thesis but I cant seem to figure something out ..

michalmutka, 2021-04-01:
> Just to add, we bought new SX1276RF1IAS module to make sure we didn't burn the modules accidentally . We got the same results. the best RSSI I achieved in about 1 meter distance between modules was [...]

I found you were commenting this post about SX1276RF1IAS. Would you be willing to answer some questions
regarding this module? I am working on my bachelor thesis but I cant seem to figure something out ..

HamedJafarzadeh, 2021-04-01:
> I found you were commenting this post about SX1276RF1IAS. Would you be willing to answer some questions
> regarding this module? I am working on my bachelor thesis but I cant seem to figure something [...]

Hi Michal,
Sure, feel free to send your issues here and lets see how others or I can help you .

michalmutka, 2021-04-01:
> Hi Michal,
> Sure, feel free to send your issues here and lets see how others or I can help you .

Thanks!
Is there a possibility to write you email or something more personal?
Thank you!

harun2301, 2023-10-14:
Hi, I've been strugling using this LoRa library. My doubts are about setTxPower() function.
When using PA BOOST why 18-20 es mapped to 15-17? Wy can't I reach 20 dBm power output?