sandeepmistry/arduino-LoRa #49: Send struct

spa-sam, 2017-10-09:
Can you add an example of how to send and receive a structure?

zybernatic, 2017-10-10:
I also got the same issue about sending the packet. it seem that it stuck in this line

`while((readRegister(REG_IRQ_FLAGS) & IRQ_TX_DONE_MASK) == 0);`

it keep looping forever.

i use the board from heltec

and here is my sketch

`#include <SPI.h>
#include <LoRa.h>
#include<Arduino.h>

// WIFI_LoRa_32 ports

// GPIO5  -- SX1278's SCK
// GPIO19 -- SX1278's MISO
// GPIO27 -- SX1278's MOSI
// GPIO18 -- SX1278's CS
// GPIO14 -- SX1278's RESET
// GPIO26 -- SX1278's IRQ(Interrupt Request)

#define SS      18
#define RST     14
#define DI0     26
#define BAND    433E6  //915E6 -- 这里的模式选择中，检查一下是否可在中国实用915这个频段

int counter = 0;

void setup() {
  pinMode(25,OUTPUT); //Send success, LED will bright 1 second
  
  Serial.begin(115200);
  while (!Serial); //If just the the basic function, must connect to a computer
  
  SPI.begin(5,19,27,18);
  LoRa.setPins(SS,RST);
  LoRa.setTxPower(10);
//  Serial.println("LoRa Sender");

  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  Serial.println("LoRa Initial OK!");
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;
  digitalWrite(25, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(25, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  
  delay(3000);
}`

sandeepmistry, 2017-11-13:
@spa-sam you can use https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#writing but pass in the address and length of the structure, you'll have to cast to `uint8_t*`.

closed 2017-11-13 by sandeepmistry

richardfyeo, 2018-04-21:
Hi,
Just in case anyone else is still wondering about this question, as Sandeep mentions above, you can read and write using the address of the buffer/structure etc that you have filled with data.   
So for the Tx you have 
```
LoRa.beginPacket();
LoRa.write((uint8_t*)&data, sizeof(data));
LoRa.endPacket();
```
and for the Rx I have 
 ```
int packetSize = LoRa.parsePacket ();
 if (packetSize) // Only read if there is some data to read..
  {
    LoRa.readBytes((uint8_t *)&data, packetSize);
  }
```
This is working fine for me using a structure containing a mix of floats, ints and bytes, although I got caught out by the different default sizes of ints on ESP32 (Rx) and the ATmega (Tx). On the ESP32 I had to explicitly define the ints in my data structure as short ints, otherwise the int values received were wrong..

akshay0070, 2020-06-02:
if I want to send an acknowledgement back to the sender after receiving the data, how to establish an acknowledgement system. 
tried the above-mentioned code by Richardfyeo that is not working for me can you just explain little more

IoTThinks, 2020-06-02:
@akshay0070 Start a new thread.

akshay0070, 2020-06-03:
I think time mismatch is happening that's why unable to receive an acknowledgement from the receiver
any help will be appreciated