sandeepmistry/arduino-LoRa #52: Bad LoRa Range with electron Particle.

pearteardial, 2017-10-20:
Hi Everyone,

I am using an electron with a SX1276 lora module to act as a receiver and I'm having a strange issue.

When I connect up the LoRa module to a Arduino mini, I get good range (100m in built up city).

When I connect up the electron to the same LoRa module the range drops to about 25m.
any idea why this might be?

Here are the reg dumps from both setups: 

> ok here's the arduino 
> LoRa Dump Registers
> 0x0: 0x44
> 0x1: 0x81
> 0x2: 0x1A
> 0x3: 0xB
> 0x4: 0x0
> 0x5: 0x52
> 0x6: 0xE4
> 0x7: 0xC0
> 0x8: 0x0
> 0x9: 0x8F
> 0xA: 0x9
> 0xB: 0x2B
> 0xC: 0x23
> 0xD: 0x2
> 0xE: 0x0
> 0xF: 0x0
> 0x10: 0x0
> 0x11: 0x0
> 0x12: 0x0
> 0x13: 0x0
> 0x14: 0x0
> 0x15: 0x0
> 0x16: 0x0
> 0x17: 0x0
> 0x18: 0x10
> 0x19: 0x0
> 0x1A: 0x0
> 0x1B: 0x0
> 0x1C: 0x0
> 0x1D: 0x72
> 0x1E: 0x70
> 0x1F: 0x64
> 0x20: 0x0
> 0x21: 0x8
> 0x22: 0x1
> 0x23: 0xFF
> 0x24: 0x0
> 0x25: 0x0
> 0x26: 0x4
> 0x27: 0x0
> 0x28: 0x0
> 0x29: 0x0
> 0x2A: 0x0
> 0x2B: 0x0
> 0x2C: 0x0
> 0x2D: 0x50
> 0x2E: 0x14
> 0x2F: 0x45
> 0x30: 0x55
> 0x31: 0xC3
> 0x32: 0x5
> 0x33: 0x27
> 0x34: 0x1C
> 0x35: 0xA
> 0x36: 0x3
> 0x37: 0xA
> 0x38: 0x42
> 0x39: 0x12
> 0x3A: 0x49
> 0x3B: 0x1D
> 0x3C: 0x0
> 0x3D: 0xAF
> 0x3E: 0x0
> 0x3F: 0x0
> 0x40: 0x0
> 0x41: 0x0
> 0x42: 0x12
> 0x43: 0x24
> 0x44: 0x2D
> 0x45: 0x0
> 0x46: 0x3
> 0x47: 0x0
> 0x48: 0x4
> 0x49: 0x23
> 0x4A: 0x0
> 0x4B: 0x9
> 0x4C: 0x5
> 0x4D: 0x84
> 0x4E: 0x32
> 0x4F: 0x2B
> 0x50: 0x14
> 0x51: 0x0
> 0x52: 0x0
> 0x53: 0x11
> 0x54: 0x0
> 0x55: 0x0
> 0x56: 0x0
> 0x57: 0xF
> 0x58: 0xE0
> 0x59: 0x0
> 0x5A: 0xC
> 0x5B: 0x0
> 0x5C: 0x7
> 0x5D: 0x0
> 0x5E: 0x5C
> 0x5F: 0x78
> 0x60: 0x0
> 0x61: 0x1C
> 0x62: 0xE
> 0x63: 0x5B
> 0x64: 0xCC
> 0x65: 0x0
> 0x66: 0x1
> 0x67: 0x50
> 0x68: 0x0
> 0x69: 0x0
> 0x6A: 0x0
> 0x6B: 0x0
> 0x6C: 0x0
> 0x6D: 0x0
> 0x6E: 0x0
> 0x6F: 0xB
> 0x70: 0xD0
> 0x71: 0x0
> 0x72: 0x14
> 0x73: 0x0
> 0x74: 0x0
> 0x75: 0x0
> 0x76: 0x0
> 0x77: 0x0
> 0x78: 0x0
> 0x79: 0x0
> 0x7A: 0x0
> 0x7B: 0x0
> 0x7C: 0x0
> 0x7D: 0x0
> 0x7E: 0x0
> 0x7F: 0x0

> Particle & LoRa
> LoRa Dump Registers
> 0x0: 0x49
> 0x1: 0x81
> 0x2: 0x1A
> 0x3: 0xB
> 0x4: 0x0
> 0x5: 0x52
> 0x6: 0xE4
> 0x7: 0xC0
> 0x8: 0x0
> 0x9: 0x8F
> 0xA: 0x9
> 0xB: 0x2B
> 0xC: 0x23
> 0xD: 0x1
> 0xE: 0x0
> 0xF: 0x0
> 0x10: 0x0
> 0x11: 0x0
> 0x12: 0x0
> 0x13: 0x0
> 0x14: 0x0
> 0x15: 0x0
> 0x16: 0x0
> 0x17: 0x0
> 0x18: 0x10
> 0x19: 0x0
> 0x1A: 0x0
> 0x1B: 0x0
> 0x1C: 0x0
> 0x1D: 0x72
> 0x1E: 0x70
> 0x1F: 0x64
> 0x20: 0x0
> 0x21: 0x8
> 0x22: 0x1
> 0x23: 0xFF
> 0x24: 0x0
> 0x25: 0x0
> 0x26: 0x4
> 0x27: 0x0
> 0x28: 0x0
> 0x29: 0x0
> 0x2A: 0x0
> 0x2B: 0x0
> 0x2C: 0x0
> 0x2D: 0x50
> 0x2E: 0x14
> 0x2F: 0x45
> 0x30: 0x55
> 0x31: 0xC3
> 0x32: 0x5
> 0x33: 0x27
> 0x34: 0x1C
> 0x35: 0xA
> 0x36: 0x3
> 0x37: 0xA
> 0x38: 0x42
> 0x39: 0x12
> 0x3A: 0x49
> 0x3B: 0x1D
> 0x3C: 0x0
> 0x3D: 0xAF
> 0x3E: 0x0
> 0x3F: 0x0
> 0x40: 0x0
> 0x41: 0x0
> 0x42: 0x12
> 0x43: 0x24
> 0x44: 0x2D
> 0x45: 0x0
> 0x46: 0x3
> 0x47: 0x0
> 0x48: 0x4
> 0x49: 0x23
> 0x4A: 0x0
> 0x4B: 0x9
> 0x4C: 0x5
> 0x4D: 0x84
> 0x4E: 0x32
> 0x4F: 0x2B
> 0x50: 0x14
> 0x51: 0x0
> 0x52: 0x0
> 0x53: 0x11
> 0x54: 0x0
> 0x55: 0x0
> 0x56: 0x0
> 0x57: 0xF
> 0x58: 0xE0
> 0x59: 0x0
> 0x5A: 0xC
> 0x5B: 0x0
> 0x5C: 0x7
> 0x5D: 0x0
> 0x5E: 0x5C
> 0x5F: 0x78
> 0x60: 0x0
> 0x61: 0x1C
> 0x62: 0xE
> 0x63: 0x5B
> 0x64: 0xCC
> 0x65: 0x0
> 0x66: 0x1
> 0x67: 0x50
> 0x68: 0x0
> 0x69: 0x0
> 0x6A: 0x0
> 0x6B: 0x0
> 0x6C: 0x0
> 0x6D: 0x0
> 0x6E: 0x0
> 0x6F: 0xB
> 0x70: 0xD0
> 0x71: 0x0
> 0x72: 0x13
> 0x73: 0x0
> 0x74: 0x0
> 0x75: 0x0
> 0x76: 0x0
> 0x77: 0x0
> 0x78: 0x0
> 0x79: 0x0
> 0x7A: 0x0
> 0x7B: 0x0
> 0x7C: 0x0
> 0x7D: 0x0
> 0x7E: 0x0
> 0x7F: 0x0

dc2mw, 2017-11-12:
My advice is to go for some sort of RF measurement equipment. For 20 US$ you can setup a software defined radio (RTL-SDR dongle + piece of wire as antenna + software of choise, e. g. GNU Radio, SDR#, GQRX etc. ) and "measure" or at least check RF output signals of modules.

Then you can see if the library configures your specific modules correctly, e. g. Reg 0x01 Bit3, Reg 0x09 Bit7 and Reg 0x4d (0x84 vs. 0x87) must be correctly set for the brand/version of the module. 
Not simply equally set for any SX1276 chip! 
The difference between correct/incorrect settings is huge and can be observed with low-cost RF measurement equipment. 

LoRa is problematic in this respect, even in case of severe mistakes/faulty modules one might still get hundred meters of range :-)

sandeepmistry, 2017-11-13:
@pearteardial are there any differences in the register values between the two?

Maybe the Particle board can't supply enough current, can you try with an external 3.3V source?

sandeepmistry, 2017-12-18:
Closing this for now due to lack of feedback, please re-open with requested info. if you are still interested.

closed 2017-12-18 by sandeepmistry