sandeepmistry/arduino-LoRa #56: saving pins ss and reset

universam1, 2017-11-04:
Pins on ESP8266 platform are expensive. 
Reading through your great library I assume that the pins ss and reset could be omitted. 

SS could be pulled down permanently if single SPI device. 
Reset could be wired to ESP8266 reset pin and get resetted at boot. 
A little support by library would be required I could provide and test. 
What do you think about?

sandeepmistry, 2017-11-13:
@universam1 please submit a pull request to continue the discussion.

universam1, 2017-11-13:
Roger, I will test first, ordered hardware.

universam1, 2017-11-16:
Okay I tested this whole setup, and must step back from suggesting to save pin `ss`, it does not work in my setup ESP8266 with RA-02 reliable, so removed it from MR.

Nevertheless, omitting the `reset` pin by hardwiring to the MCU works just perfect, for me there is no use to spend an IO for this. 
Feel free to merge @sandeepmistry

sandeepmistry, 2017-12-18:
Closed via #60.

closed 2017-12-18 by sandeepmistry