sandeepmistry/arduino-LoRa #63: Non blocking function to transmit

universam1, 2017-11-17:
The `endPacket()` is a blocking transmit that can take over 6 seconds depending on the radio settings. 
A non blocking version is proposed with MR #62 , tested successfully on ESP8266. 
Note the grace time required for the radio to properly update the registers.

Example added as well to demo.

ainst, 2018-02-25:
Would it be possible to add a callback as well as part of the changes to endPacketAsync()? 

For example, I would like to turn receive back on once a transmit has been completed.

torntrousers, 2018-04-07:
Agree with @ainst , a callback for tx done would make this async support much more useful.

sandeepmistry, 2018-08-19:
Closed via #62.

> Would it be possible to add a callback as well as part of the changes to endPacketAsync()?

@ainst @torntrousers  please open a new pull request for this.

closed 2018-08-19 by sandeepmistry