sandeepmistry/arduino-LoRa #64: problem while sending messages

halukmy, 2017-11-19:
thanks for code

i just got some problem its like

i made one device is gateway and others are nodes

for example i send %20 all nodes, and
nodes replies i'm alive

but while same time message, message cant recieve

is there any fix or way?

thanks

sandeepmistry, 2017-11-21:
@halukmy Can you please provide example sketches?

halukmy, 2017-12-15:
@sandeepmistry for example i just made stress test, on loop send messages,

one is higher other is lover

sandeepmistry, 2017-12-18:
@halukmy please paste the exact sketches you are using here, this way there is no guessing for how to reproduce the issues you are seeing.

halukmy, 2017-12-18:
i will try to be simple

i got two esp12e device and one uno they has all drf1278f 

one is gateway, collect the data also send transmission

in case in other one esp12e and uno on loop stage sending messages repeatly

i see one of sensor works higher than other but when i close energy of it, it keeps same cycle

i wonder,

is it possible work with 200 sensors? gateway can collect? 

did you try that max sensors with lora?

thanks

sandeepmistry, 2018-01-08:
Closing this for now, please provide a sketch to reproduce the original topic.

closed 2018-01-08 by sandeepmistry