sandeepmistry/arduino-LoRa #68: Starting LoRa failed!

ioeides, 2017-12-03:
Hey. I have a problem with my LoRa node. It cosists of a PCB designed by myself that conntcts a RFM95W with an Ardunio Pro Mini. The setup runs on an external 3v3 power supply.

Connected Pins are:
```
MISO_MISO
MOSI_MOSI
SCK__SCK
NSS__D15
RST__D14
DIO0_D8
DIO1_D9
DIO2_D10
```

Now, the Serial Prompt says:
```
LoRa Sender
Starting LoRa failed!
```

I use the Arduino IDE Version 1.8.5 and the "LoRaSender" Sketch from the Examples with some changes (bold).
```
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  **LoRa.setPins(15, 14, 8);**
  **LoRa.setSPIFrequency(1e6)**

  if (!LoRa.begin(**858E6**)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```
I have triple-checked my PCB layout and power supply. Is this a software issue?

ioeides, 2017-12-03:
BTW: My DIO0-Pin is not Interrupt-capable but that shouldn't be the issue as this affects TX-mode only, right?

3Dtj, 2017-12-15:
Hey. Did you figure out the problem to the error?  I'm still having problems getting mine to initialize.  Suggestions?

ioeides, 2017-12-15:
No. I still have the problem.

sandeepmistry, 2017-12-18:
@ioeides if you add a `Serial.println(version)` here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L87 - what is the value?

ioeides, 2017-12-18:
Changing

```
  uint8_t version = readRegister(REG_VERSION);
  if (version != 0x12) {
    return 0
  }
```

to

```
  uint8_t version = readRegister(REG_VERSION);
  if (version != 0x12) {
    return Serial.println(version);
  }
```

the serial plot says:

```
LoRa Sender
0
Sending packet: 0
```

Felix1954, 2017-12-31:
Same problem, resolved adding a 10ms delay in library.
int LoRaClass::begin(long frequency)
...
// set SS high
  digitalWrite(_ss, HIGH);
  delay(10); // add
...

ioeides, 2018-01-01:
> Same problem, resolved adding a 10ms delay in library.
This does not work for me. Still the same problem. I tried with different hardware.

MauriPastorini, 2018-01-02:
Having the same problem here

MauriPastorini, 2018-01-02:
> Sandeepmistry:
> if you add a Serial.println(version) here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L87 - what is the value?
>

I am also having the "0" version, What does it mean? How can I get the 0x12 that it is expecting?
Best regards

MauriPastorini, 2018-01-02:
@sandeepmistry Please! I will star your repoo

KE6MTO, 2018-01-04:
I'm having the same issue right now. I have a RA-01 module that I'm attempting to use with a Particle Proton. What is happening is during setup, it's checking the "Revision" register (0x42) and based on the datasheet should get 0x12 back. He's using that as a test to ensure proper communication to the chip. Right now with my Saleae I can see the 0x42 byte going back and either nothing or a random response. What i'm suspecting at the moment is the lack of enough 3.3 power so that will be my next test.

sandeepmistry, 2018-01-08:
@MauriPastorini please see https://github.com/sandeepmistry/arduino-LoRa#faq.

@ioeides 

> What does the "0" mean in my serial promt mean?

This means there is something wrong with the SPI connection or CS pin.

I'm going to close this for now, as this seems like hardware related issues. Lots of people have this library functioning.

closed 2018-01-08 by sandeepmistry