sandeepmistry/arduino-LoRa #72: Bad RSSI

DussanR, 2017-12-05:
When sending in a room no more than 2 meters apart, i get a RSSI of around 70. In the backyard through a couple of walls and 20 meters tops a RSSI of 105.

I wanted to make this better, but i don't quite understand how to use the setTxPower. If someone could help me i would really appreciate it!

dontsovcmc, 2017-12-16:
I use https://www.ebay.com/itm/LoRa32u4-II-Lora-LiPo-Atmega32u4-SX1276-HPD13-868MHZ-EU-Antenna/282623601184 with external antennas 20cm. 
I see very strange RSSI -115 when distance is 1 meter 👍 (
LoRa.setTxPower(17);
LoRa.setSignalBandwidth(31.25E3); /
LoRa.setSpreadingFactor(7); 
 LoRa.setCodingRate4(5); 

Is it normal?

DussanR, 2017-12-18:
I am using a dragino shield on an arduino uno. I have the antenna the shield came with. 

Would it be something like ... ?

`LoRa.setTxPower(17, PA_OUTPUT_PA_BOOST_PIN);`

Thanks!

dontsovcmc, 2017-12-20:
For my problem: I can't find the reason ( Sometimes RSSI is -70, sometimes -20 (normal) for 30cm.

sandeepmistry, 2018-01-08:
@DussanR `LoRa.setTxPower(17)l` or `LoRa.setTxPower(17, PA_OUTPUT_PA_BOOST_PIN);` is fine.

Do you have a photo of the shield? Are you passing in the correct frequency into begin that the board and antenna are designed for?

DussanR, 2018-01-08:
This is the shield:

http://wiki.dragino.com/index.php?title=File:Lora_Shield_v1.4.jpg

With the second parameter do i have to declare a new pin? 

And yes, i have the shield/antenna for 915 and using the LoRa.begin(915E3)!

sandeepmistry, 2018-01-08:
> With the second parameter do i have to declare a new pin?
Nope.

I have the same shield here, and am getting reasonable RSSI.

What board are you connecting the shield too?

DussanR, 2018-01-08:
To an original arduino. Odd, what could be the problem?

sandeepmistry, 2018-01-08:
Could be the 3.3V doesn't have enough current or power source generating noise?

I've been using and Uno with the shield.

DussanR, 2018-01-08:
I’ll try using the second parameter, cause i’ve used different arduino unos with the same problem. 

Thanks!

boonkerz, 2018-01-16:
i have the same issue only when i place both near by i can transmit and receive messages.
Arduino UNO with DRF1278F

Sukuri, 2018-02-03:
Hi, I'm using arduino uno with RFM95 915MHz, can I use LoRa.setTxPower(17, PA_OUTPUT_PA_BOOST_PIN); as well?
Is it really limited to 17 because from datasheet maximum is +20dbm,
and the pin called PA_BOOST, or just use PA_OUTPUT_PA_BOOST_PIN?

dontsovcmc, 2018-02-03:
I see RSSI -20 with SX1276 ~0.5m. 
I think LoRa.setTxPower(17, PA_OUTPUT_PA_BOOST_PIN);  is not needed, cause it has been already called somewhere (check LoRa.begin())

Sukuri, 2018-02-04:
@dontsovcmc thanks. But I've still no idea about my RSSI, I only can get -20 for a few centimeters. 
What SF you are using?

Sukuri, 2018-02-06:
@sandeepmistry 
I just need to check with you about REG_PKT_RSSI_VALUE register you're using is 0x1a right? 

when I check for sx1276/7/8 here: https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf , it's correct, 0x1a is used.

but when I check for RFM95/6/7/8 here: http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf ,
it using is 0x24.

Correct me if I'm wrong. Thanks.

dontsovcmc, 2018-02-06:
check https://www.elecrow.com/wiki/index.php?title=Lora_RFM95_Shield-_915MHZ
(use another RadioHead library example)
or check RSSI with:
setTxPower(14, PA_OUTPUT_RFO_PIN) for both modules & check RSSI
setTxPower(14) for both modules & check RSSI

Sukuri, 2018-02-07:
@dontsovcmc thanks for that. I've test it but still low RSSI. These are my current settings:

  LoRa.setSpreadingFactor(12);       
  LoRa.setSignalBandwidth(125E3); 
  LoRa.setTxPower(17, PA_OUTPUT_PA_BOOST_PIN);
  LoRa.setCodingRate4(5);           
  LoRa.setSyncWord(0xF3);           
  LoRa.setPreambleLength(6);
  LoRa.enableCrc();

I'm not sure, maybe it's from my hardware problem. By the way, thanks again :)

halukmy, 2018-02-07:
@boonkerz  i got that error before, just change 868E6 to 433E6 ;) it will rock'n roll

KE6MTO, 2018-02-07:
The antenna may be questionable. May not be the same boards, but here is a youtube video of someone who tested the antenna's that came with some of the cheep LoRa boards: https://www.youtube.com/watch?v=CJNq2I_PDHQ 

Needless to say, they weren't good...

I got new antenna's from Linx (purchased via Mouser): https://linxtechnologies.com/wp/p/antennas/small-external-antennas/

cassiolucas, 2018-02-23:
Hello guys,

I have the same issue, currently I am using arduino uno and NiceRF1276 lora modules on both ends, but, the RSSI value is around -60 and both arduino are just side-by-side...

Did any of you find a solution for this?

sandeepmistry, 2018-02-26:
@cassiolucas please see https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power maybe you need to try an additional parameter?

sandeepmistry, 2018-02-26:
@DussanR any progress on this?

dontsovcmc, 2018-02-26:
@cassiolucas are you choose frequency right? :/ 
@sandeepmistry what parameters can you offer?!

DussanR, 2018-02-26:
@sandeepmistry i added the second parameter and it's basically the same. i get 60-70 in the same room, and around 100 in my backyard

cassiolucas, 2018-02-27:
Hello @dontsovcmc, yes the right frequency 915Mhz.

@sandeepmistry I believe this cannot be improved, I have tested a different library for arduino and the RSSI was only around 10 dBm better... 

@DussanR Yes I have also done a lot of tests with the parameters from API.md but still the same, I just dont understand why its said that this module can work in long range (2km between buildings) but on my tests I didnt get even close to 1km.

dontsovcmc, 2018-02-27:
@cassiolucas check RF settings for better distance. You should choose good antenna!
I got 1 km through buildings with very fast (bad) settings with +5dBm 20cm antennas:
power: 20 dBm, 62.5kHz, SF 8, CR 4/6
-20 RSSI ~ 5 sm
-50 RSSI ~ 5 meters.

cassiolucas, 2018-02-27:
@dontsovcmc can you please share your code and how were you able to set the +20dBm power?

The Lora module and antenna which I am currently using is:
[image]

dontsovcmc, 2018-02-27:
@cassiolucas check this commit for +20dBm https://github.com/sandeepmistry/arduino-LoRa/pull/79
I cant' understand, you got good RSSI or not? 
Remember, that library for chip, but you bought a small board with chip, so API can be little different. For example, output pin. I can't understand why NiceRF connect digital IO pins to MCU.

Please check modules with home library (lora_slave, lora_master):
http://www.nicerf.com/product_146_136.html, password "nicerf" works (=)).

If success, try to understand differences with sandeepmistry library...

cassiolucas, 2018-02-27:
Hello @dontsovcmc Ok I will check the NiceRF examples codes.

Do you know how can I get the RSSI values using the NiceRF codes?

cassiolucas, 2018-02-28:
I have found how to read the RSSI and just by uploading the TX and RX code from Nicerf I get around +47 dBm as RSSI when the node is just 1 meter away.. I wasn't expecting a positive value for RSSI...Or maybe I am not reading it correctly from the library?

closed 2018-08-19 by sandeepmistry

sr58977, 2019-02-07:
@cassiolucas  hiii...
can you please mention the code for RSSI from nice rf library, it will glade to you...

thank you

AsyDynamics, 2019-04-10:
> @DussanR what hardware are you using?
> 
> Do you have the correct antenna's for the frequency you are using?
> 
> https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power
> 
> For AI thinker modules RFO is used for power instead of PA boost. so you could try:
> 
> ```
> LoRa.setTxPower(14, PA_OUTPUT_RFO_PIN);
> ```

Hi, can you give little bit more explanation of the RFO mode of ai thinker module, or is there any link because I cannot find on google. Wondering if what you said means the max power of ai thinker module is +14 ? Thanks