sandeepmistry/arduino-LoRa #74: Local address for LoRa protocol

3Dtj, 2017-12-09:
Totally new to LoRa protocols so please excuse the basic ignorance with this question.  For the Sandeep LoRa protocol, I've entered the protocol but get an error when compiling.  Issue seems to be providing the correct local address.  Is that a arbitrary value that I provide or is it specific to the LoRa radio chip?  I was under the impression that I could just provide an arbitrary value but clearly the Ardunio sketch wants something more specific.  I'm sure this is a basic operation (like pushing the call button on a phone to make a call) but I'm new to this.

Much appreciate that help.
Travis

3Dtj, 2017-12-14:
Ok I'm fairly certain the address designation is subjective thus 0xBB is perfectly fine to use for one of the radios.  But, now I'm getting the error message that initialization failed.  Wiring seems easy enough thus is there some other additional step to get radio to initialize??

[image]

3Dtj, 2017-12-15:
Shouldn't I see an LED activated on the LoRa chip regardless if the code is uploaded or not?  Seems like power to the chip should trigger some activity (am I wrong)?

sandeepmistry, 2017-12-18:
@3Dtj LoRa doesn't have any address by default, the Duplex example adds an address layer instead the LoRa packet itself.

It doesn't look like the wiring you are uses matches: https://github.com/sandeepmistry/arduino-LoRa#semtech-sx1276777879-wiring

You can use `LoRa.setPins(...)` to chance the pins: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-pins

Also you can try to lower the SPI frequency: https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

3Dtj, 2017-12-19:
Apologize for bending the rules on this.  I haven't used Github before and noticed you hadn't provided updates to the other postings thus I assumed that you were no longer availabe for assistance.

I am trying your suggestions now and will give you an update in the next hour.

Thank you,
3Dtj

3Dtj, 2017-12-19:
Below is the sketch from the Making Things Talk 3rd edition by Tom Igoe who references your code.  I've entered it into an Ardunio UNO.  I used the same wiring diagram that is illustrated in his book thus I'm confused by your wiring recommendation.

LoRa Duplex communication

  Modified code for Duplex communication.  
  Sends a message every half second, and polls continually
  for new incoming messages. Implements a one-byte addressing scheme,
  with 0xFF as the broadcast address.

  Uses readString() from Stream class to read payload. The Stream class'
  timeout may affect other functuons, like the radio's callback. For an

  created 28 April 2017 
  by Tom Igoe 
*/
#include <SPI.h>              // include libraries
#include <LoRa.h>

const int buttonPin = 4;
const int receiveLED = 5;        
const int csPin = 7;          // LoRa radio chip select
const int resetPin = 6;       // LoRa radio reset
const int irqPin = 2;         // interrupt request pin

int lastButtonState = HIGH;   //initial pushbutton state
byte msgCount = 0;            // count of outgoing messages
byte localAddress = 0xBB;     // address of this device
byte destination = 0xFF;      // destination to send to
byte syncWord = 0xB4;         // Sync word (network ID)
byte spreadingFactor = 7;     //spreading factor (6-12)

void setup() {
  Serial.begin(9600);                     // initialize serial
Serial.println("LoRa Duplex");
  LoRa.setPins(csPin, resetPin, irqPin);  // set CS, reset, IRQ pin
Serial.println("LoRa Duplex");
  if (!LoRa.begin(915E6)) {               // initialize ratio at 915 MHz
    Serial.println("LoRa init failed. Check your connections.");
    while (true);                         // if failed, do nothing
  }
LoRa.setSyncWord(syncWord);
LoRa.setSpreadingFactor(spreadingFactor);
LoRa.setTimeout(10);                      //set Stream timeout of 10ms
Serial.println("LoRa init succeeded.");   //set the I/O pin modes:
pinMode(buttonPin,  INPUT_PULLUP);
pinMode(receiveLED,   OUTPUT);
}

void loop() {
  int buttonState = digitalRead(buttonPin); // read the pushbutton
  if (buttonState  != lastButtonState){          // if it's changed
    delay (3);                              // debounce delay
    if (buttonState  ==LOW)  {              // if it's low
      String message = "HeyLoRa!";           // send a message
      sendMessage(message);
      Serial.println(message);              // print the message sent
    }
    lastButtonState = buttonState;          // save button state
  }
  
  // parse for a packet, and call onReceive with the result:
  onReceive(LoRa.parsePacket());
}

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

digitalWrite(receiveLED, HIGH);         // turn on the receive LED
  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte msgLength = LoRa.read();    // incoming msg length

  String incoming = LoRa.readString();  //payload of packet

  if (msgLength != incoming.length()) {  // check length for error
    Serial.println("error: message length is wrong");
    return;                             // skip rest of loop
  }

  // if the recipient isn't this device or broadcast,
  if (recipient != localAddress && recipient != 0xFF) {
    Serial.println("This message is not for me.");
    return;                             // skip rest of loop
  }

  // if message is for this device, or broadcast, print details:
  Serial.print("Received from: ");
  Serial.println(sender, HEX);
  Serial.print("Sent to:  ");
  Serial.println(recipient, HEX);
  Serial.print("Message ID: ");
  Serial.println(incomingMsgId);
  Serial.print("Message length: ");
  Serial.println(msgLength);
  Serial.print("Message: ");
  Serial.println(incoming);
  Serial.print("RSSI: ");
  Serial.println(LoRa.packetRssi());
  Serial.print("Snr: ");
  Serial.println(LoRa.packetSnr());
  Serial.println();
  digitalWrite(receiveLED,  LOW);     // turn off receive LED

3Dtj, 2017-12-19:
[image]

This is the wiring diagram as illustrated in the book

3Dtj, 2017-12-19:
1.8,2 is my Arduino IDE

3Dtj, 2017-12-19:
[image]

3Dtj, 2017-12-19:
[image]

3Dtj, 2017-12-19:
When I run the sketch, I do get the two LoRa Duplex messages on the serial monitor and the LoRa init failed. Check your connections.  Thus the sketch compiles and is sending it to the UNO but not sure why the fail message especially since I have the wiring mimicing the book's illustration.

3Dtj, 2017-12-20:
Also,  is the below code common for sending messages to radio chips?

LoRa.beginPacket(); // start packet

3Dtj, 2017-12-23:
Sandeep, any suggestions?

3Dtj, 2017-12-23:
Confirmed that I have cpp and h files in my library folder.

3Dtj, 2018-01-01:
Any suggestions?   Am I missig something obvious or should I use a different sketch/code?

Thank you,
Travis

sandeepmistry, 2018-01-08:
@3Dtj your Uno (clone) board operates at 5V not 3.3V, I've updated the read me to include a note about this https://github.com/sandeepmistry/arduino-LoRa/commit/77985324358ebcf7cd4c96feb8b5e109f5d914b7. The Arduino 101 in the book operates at 3.3V.

Luckily the Adafruit breakout board you are using has a built-in level converter, so you just need to connect the Vin pin to 5V instead or 3.3V and things should be fine.

@tigoe if the 5V vs 3.3V is not mentioned in the book, it would be good to add as an errata and update for next revision.

closed 2018-01-08 by sandeepmistry

tigoe, 2018-01-08:
The diagram in the book referenced runs the board from the 3.3V rather than the 5V, but I'll add it to the list to mention explicitly should there be a future edition.

3Dtj, 2018-01-08:
I just saw that you suggested that to another user too.  I'll try that out
and respond back.