sandeepmistry/arduino-LoRa #75: $50 to get this code working for two RFM95W LoRa chips

3Dtj, 2017-12-15:
Hoping the incentive of $$ will prompt somebody to respond and help me get this working.  No scam here.  $50 is legitimate to anybody out there that provides me tips to get this code working using two Arduino UNOs and two RFM95W LoRa radio chips.  Deal is good until December 24 2017.

sagram, 2017-12-15:
what exactly you want

sabas1080, 2017-12-15:
I have a example but other library and work https://github.com/ElectronicCats/LoraShield/tree/master/Examples/LoRa_PHY

sagram, 2017-12-15:
check mine post before couple of days ago , it's sender and work normally. all library work well. if you want for money make upwork job and we can make it

sagram, 2017-12-15:
one more thing , there is frequency issues, some china work on 433 so put 433E6 in LoRa.begin.... or what frequency is chip

sabas1080, 2017-12-15:
RFM95 work to 915 and 868Mhz

wero1414, 2017-12-15:
Hi 3Dtj, how are you doing the conection to the chip? by a shield? or by a breadboard? and would you share your project? it would be usefull to know how to proceed with the code

3Dtj, 2017-12-17:
Thanks all!  I'll check code /tips.  working thus need another day.

sandeepmistry, 2017-12-18:
@3Dtj please avoid create multiple issues and posting "bounties" in this issue tracker.

I've replied to your original post in https://github.com/sandeepmistry/arduino-LoRa/issues/74#issuecomment-352304975.

closed 2017-12-18 by sandeepmistry