sandeepmistry/arduino-LoRa #77: Support +20 dBm for SX1276

dontsovcmc, 2017-12-17:
It's will be good to add +20 dBm mode for SX1276 board and written in manual http://www.semtech.com/images/datasheet/sx1276.pdf
```
see 5.4.3. High Power +20 dBm Operation
```
I have 2 boards: https://www.ebay.com/itm/LoRa32u4-II-Lora-LiPo-Atmega32u4-SX1276-HPD13-868MHZ-EU-Antenna/282623601184

```
settings:
    LoRa.setSignalBandwidth(31.25E3); 
    LoRa.setSpreadingFactor(12); 
    LoRa.setCodingRate4(5);
    LoRa.enableCrc();
```

I check board current when transmitting at 17dBm and 20 dBm settings (as read in manual):

17 dBm:
` LoRa.setTxPower(17);`
current receive mode: 23mA
current transmit 1 byte with crc: 66mA (~4sec), 73mA (~2 sec)

20 dBm:
```
    writeRegister(REG_PA_CONFIG, PA_BOOST | (17 - 2));
    writeRegister(REG_PA_DAC, 0x87);  //turn on 3rd amplifier, see 5.4.3
    writeRegister(REG_OCP, 0x20 | 18); //increace power protection to 150mA RegOcp, see 5.4.4
```
current receive mode: 23mA
current transmit 1 byte with crc: 92mA (~4sec), 98mA (~2 sec)

I also see ~2 RSSI difference on receiver between 17 and 20 modes. 
I hadn't measure distance, sorry.

p.s. to turn off 20 dBm call:
```
    writeRegister(REG_PA_DAC, 0x84); //turn off 3rd amplifier, see 5.4.3
    writeRegister(REG_OCP, 0x20 | 0x0B); //RegOcp default
```

sandeepmistry, 2017-12-18:
Hi @dontsovcmc, please submit a pull request for the suggested changes.

Sukuri, 2018-02-05:
@dontsovcmc is this can be used in RFM96? and can I know what is the RFM96 PA_BOOST pin?, I'm confused about this pin because I didn't get it in datasheet

dontsovcmc, 2018-02-05:
I don't read your manual. 
 RFM96 should be the same  & pin should be soldered on board.

halukmy, 2018-02-05:
@dontsovcmc  add 1278 support  too

Sivick, 2018-03-17:
Hola buenas, estoy experimentando con este libreria y estoy midiendo el alcance maximo

Tengo el SX1276 Lora shield  --- Dragino Lora Shield  basado RF95, segun el lo que he investigado puedo colocarle hasta 20 db de potencia.

LoRa.setTxPower(20);

En este post no entiendo que hacen para no limitar a 17 db
Gracias

dontsovcmc, 2018-03-17:
What the question?
I don't test code with RF95. Please, check current before measure range.

Sivick, 2018-03-17:
Currently the arduino reaches me approximately 1 km with 17 db, my question is for the libery,  I am limited to 17 db, yes it supports me up to 20 db?
I just want to make it work a range of 2 km as the specifications of the shield

dontsovcmc, 2018-03-17:
1. LoRa range depends of radio settings:
LoRa.setSignalBandwidth(31.25E3); 
LoRa.setSpreadingFactor(12); 
LoRa.setCodingRate4(5);
2. Check your antenna DB value. There is a big difference between 9cm wire (0db) and SMA antenna (+5db). Check frequency of antenna. 433 is bad for 868/915 Mhz. Antenna from wi-fi router is bad (2.4Ghz).
After this knowledge, you should think about +20dBm.

1km is bad. I got >2km with SMA antennas +17dBm with SX1276.
I got 1km in the city througth buildings.

DaveCalaway, 2018-03-26:
I done the same test with RFM96:
- code example "sender": https://goo.gl/nvG4JC
- 8.2cm wire antenna

I read 20-25 mA during transmission. According to https://goo.gl/H9ZdmN, by default, my code was enabled to used +17dB.

But from the datasheet http://www.hoperf.com/upload/rf/RFM95_96_97_98W.pdf ( page 13 ) i should read 80/90 mA.
20 mA is the value for + 7 dBm.

Someone can help me?

dontsovcmc, 2018-03-26:
read 20-25 mA during transmission - you use simple multimeter or oscilloscope ?
20mA - is receiving current.
Increase transmitting time to see transmitting current on multimeter:
1. write a very long message
2. set: 
LoRa.setSignalBandwidth(31.25E3); 
LoRa.setSpreadingFactor(12); 

check current

DaveCalaway, 2018-03-27:
@dontsovcmc 
You are right, now i see something around 100mA.

dontsovcmc, 2018-03-27:
@DaveCalaway, yeah =)
I don't understand, you turn on +20dBm or use default +17dBm?
Please, check current with my additional code  for +20dBm! 
 https://github.com/sandeepmistry/arduino-LoRa/pull/79
Maybe you can measure range between +17dBm and +20dBm?