sandeepmistry/arduino-LoRa #80: Add function to correct frequency

dontsovcmc, 2017-12-20:
We can correct frequency, that can go away little bit depence of temperature end etc.

4.1.5. Frequency Error Indication of http://www.semtech.com/images/datasheet/sx1276.pdf

I see only one code about this feature: https://electronix.ru/forum/index.php?showtopic=140311&hl=lora
But I can't understand some variables and not test it.

LoRaTracker, 2018-01-12:
Frequency correction for received packets might be desirable if you using bandwidths of 41.7khz and lower, and you can mitigate the effects somewhat by calibrating each module. 

However, whilst you can correct for frequency drift once you have established packet reception, if the RX and TX ends are outside the capture range (25% of bandwidth) packet reception will not start.

dontsovcmc, 2018-01-12:
Transmitter can't be outdoor and receiver indoor.  It will be big temperature difference in winter.