sandeepmistry/arduino-LoRa #82: Data Rate - Receiving only 256 Bytes per second

Discoverideas, 2018-01-01:
Hi,

I am using LoRa module RA-02 and sending a 50kb file. I sending it to other side in 256 chunk. 
Issue is i can only receive 256 Bytes per sec when i can send 4 packets per sec. 

I am using the same library also not able to change bandwidth.

Thanks in advance!

sandeepmistry, 2018-01-08:
@Discoverideas please provide the sketches you are using. 

Aside, LoRa has a really low bit rate, sending a 50 kB file is probably not a great idea ...

Discoverideas, 2018-01-08:
Hi Sandeep,

Thanks for the response. We are interfacing ESP32 BLE with LoRa module. The purpose of the application is to connect ESP32 through BLE to mobile application and can send and receive data from other device on long range. 50 KB of file is basically a audio file. The system will be used as for chating and voice messaging. 

Below is the code
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleWrite.cpp
    Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <SPI.h>
#include <LoRa.h>
#include <TimeLib.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

uint8_t value = 0;
int count = 0; uint32_t totllenght = 0;
class MyCallbacks: public BLECharacteristicCallbacks {

    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string msg_rcvd = pCharacteristic->getValue();
      int total_len = msg_rcvd.length();
      Serial.print("total_len= ");
      Serial.println(total_len);
      int first = 0;
      int chunk = 255;
      /*if (last <= total_len) {
        last = total_len;
        }*/

      while (first < total_len) {
        int last = first + chunk;
        if (last > total_len) {
          last  = total_len;
        }
        Serial.println("*********");
        Serial.print("First - ");
        Serial.println(first);
        Serial.print("Last - ");
        Serial.println(last);

        Serial.print(hour());
        Serial.print(":");
        Serial.print(minute());
        Serial.print(":");
        Serial.print(second());
        Serial.print(":");
        Serial.println(millis());

        LoRa.beginPacket();

        for (int i = first; i <= last; i++)
        {
          Serial.print(msg_rcvd[i]);
          LoRa.print(msg_rcvd[i]);
        }
        LoRa.endPacket();
        Serial.println();

        first = first + chunk + 1;
        //delay(1000);
      }
[... 151 lines cut ...]
    Serial.print(":");
    Serial.print(minute());
    Serial.print(":");
    Serial.print(second());
    Serial.print(":");
    Serial.println(millis());

    LoRa.beginPacket();

    for (int i = first; i <= last; i++)
    {
      Serial.print(msg_rcvd[i]);
      LoRa.print(msg_rcvd[i]);
    }
    LoRa.endPacket();
    Serial.println();

    first = first + chunk + 1;
    delay(1000);
  }
  Serial.println("####################################");
}

If you have any suggestions regarding this please let me know.
Thanks,

sandeepmistry, 2018-01-14:
Hi @Discoverideas,

Could you please provide a more minimal set of sketches to identify the issue? I don't have the ESP32 LoRa boards you are using and it would be easier to diagnose if the sketch has less dependencies.

sandeepmistry, 2018-02-05:
Closing for now due to lack of activity.

closed 2018-02-05 by sandeepmistry