sandeepmistry/arduino-LoRa #83: LoRa.onReceive fails on Heloc_WiFi_LoRA_32 (ESP32)

qbzzt, 2018-01-09:
I have two of these boards, one a sender and one a receiver.

When I use polling with the receiver ([code is here](https://github.com/qbzzt/IoT/blob/master/201801/Hay_Bale_Fire/05_LoRa_Receive.ino)), I lose a few packets from time to time. [But when I try to use onReceive, the function is never called](https://github.com/qbzzt/IoT/blob/master/201801/Hay_Bale_Fire/LoRa_onReceive_Fail).

Any idea why? I set the IRQ to the correct pin, I'm pretty sure ([26, as shown here](https://cdn.hackaday.io/files/269911154782944/Heltec_WIFI-LoRa-32_DiagramPinout.jpg)).

Thanks, Ori

qbzzt, 2018-01-11:
Additional information. When I try to bypass the LoRa library the ISR still doesn't get called

>   attachInterrupt(digitalPinToInterrupt(LORA_SPI_PINS.irq), 
>     onReceive, 
>     CHANGE);

sandeepmistry, 2018-01-14:
Can you try this sketch:

```arduino
void setup() {
  Serial.begin(115200);

  attachInterrupt(digitalPinToInterrupt(26), onRise, RISING);
}

void loop() {
  delay(100);
}

void onRise() {
  Serial.println("onRise");
}
```

Then manually attach a wire on pin 26 and switch between GND and 3V3. If nothing prints out it's mostly likely an issue with the ESP32 Arduino core or hardware and you'll have to report it to the ESP32 Arduino core maintainers or the board manufacturer.

morganrallen, 2018-01-23:
I can confirm with this code that it is not receiving an interrupt on the pin labeled 26.

morganrallen, 2018-01-23:
Calling `pinMode(26, INPUT);` before makes it work, but `onReceive` still is not.

sandeepmistry, 2018-01-23:
@morganrallen can you please try to add `pinMode(_dio0, INPUT);` here: https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L281

If that works, please submit a pull request for the change, excellent debugging so far :)

morganrallen, 2018-01-23:
Hah, I was just trying exactly that. No such luck.

morganrallen, 2018-01-23:
I just manage a continuity test between what I think is DIO0 on the SX1276 and (physical) pin 26. I don't have the best equipment at home right now but it does appear the connection is correct.

morganrallen, 2018-01-23:
@sandeepmistry Are you testing with an SX1276 or SX1278? Maybe the register to enable interrupt is different?

sandeepmistry, 2018-01-23:
I'm using a SX1276, but the data sheet is the same for both.

Could 26 be connected to another DIO pin instead of DIO0? Would be nice to find the schematic if available.

@monteslu is this the same board you have a bunch of? https://cdn.hackaday.io/files/269911154782944/Heltec_WIFI-LoRa-32_DiagramPinout.jpg

morganrallen, 2018-01-24:
I've already started cold contacting Aliexpress vendors trying to dig up a schematic. We'll see how that goes.

I'm limited to just a multimeter right now, but confidence is pretty high the connection is complete.

morganrallen, 2018-01-24:
Wow! I got the schematic! Sadly, it only confirms the issue is not the connection. I'm working on a C implementation that will hopefully give a point of comparison.

BlueCoal69, 2018-01-31:
Adding the  pinMode(_dio0, INPUT); line fixed it for me but I'm using the TTGO (Heltec clone maybe) module from Aliexpress.  The only issue I have using the callback is with VSCode giving me out of scope errors for onreceive. When using the Arduino IDE it works fine and loads.

sandeepmistry, 2018-01-31:
@BlueCoal69 please provide a pull request for this :)

morganrallen, 2018-01-31:
This is good news, not sure why it did not work for me, hopefully you've done something different that I missed.

closed 2018-02-05 by sandeepmistry