sandeepmistry/arduino-LoRa #85: Setting the Low Datarate Optimization flag

LoRaTracker, 2018-01-12:
I have looked at your code and I cannot see where the LowDataRateOptimize flag is being set, or can be set ?

See section 4.1.1.6 of the SX127x datasheet, the flag (bit 3 of reg 0x26) needs to be set when the symbol time exceeds 16ms. 

I can describe the symptoms of not setting the flag if you want.

sandeepmistry, 2018-01-14:
@LoRaTracker agreed it's not set, could you please submit a pull request to set it "when the symbol time exceeds 16ms." as per the data sheet ?

Is there a calculation that can be used to determine this condition?

LoRaTracker, 2018-01-14:
There is a calculation yes, see 4.1.1.5 of the data sheet.

However, you do need to pull down the current set bandwidth and 
spreading factor then do the calculation.  Since there are already 
methods for setting bandwidth, spreading factor etc, it would be easier 
to just provide a method to set or clear the flag and make users aware 
of the setting. Then its up to the user to decide, the Semtech 
calculator tool will provide them the information.

Which option do you want a pull request for ?

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

sandeepmistry, 2018-01-19:
@LoRaTracker thanks,

so just to confirm, the formula is:
```
      BW
Rs = ----
     2^SF
```

If it's a simple formula like this, I think it's best for the user if `LoRa.setSignalBandwidth(...)` and `LoRa.setSpreadingFactor(...)` automatically do it.

> Since there are already methods for setting bandwidth, spreading factor etc, it would be easier
to just provide a method to set or clear the flag and make users aware of the setting.

Would there be unwanted consequences if the user turns it on when it's not needed, or it's off when it's needed?

LoRaTracker, 2018-01-19:
You do have to replace the binary value used to set the bandwith with 
the proper value such as 62500, 125000 etc.

If its not turned on and it should be, then you will get corruption in 
packets and this is more likely the longer the packets are and the more 
power you use. Although the issue with power only occurs with modules 
that dont use a TCXO, which unfortuntely is most of them.

Never tested the receiver with the flag on when it should not be.

sandeepmistry, 2018-01-22:
@LoRaTracker thanks for the info again!

Let's go for `LoRa.setSignalBandwidth(...)` and `LoRa.setSpreadingFactor(...)` automatically setting the correct value for Low Data Optimization then.

torntrousers, 2018-03-05:
FYI, I've started doing this here - https://github.com/torntrousers/arduino-LoRa/compare/master...torntrousers:issue-85-setting-ldo-flag
Haven't done any testing of it yet but let me know if there are any early review comments.

sandeepmistry, 2018-03-12:
@torntrousers nice!

Looks ok to me, only suggestion would be to have the new getters as private for now.

torntrousers, 2018-03-13:
My code doesn't work properly and I don't think I understand the datasheet on this.

Section 4.1.1.5 says the symbol rate is bandwidth /  (2 ^ spreading factor)
and 
Section 4.1.1.6 says the LDO bit should be set if the symbol rate exceeds 16.

but that seems to give me the inverse result. For example:

when SF is 11 and BW is 20.8k the LDO = 20800 / 2^11 = 10, so  10 is less than 16 so LDO = 0
when SF is 11 and BW is 62.5k the LDO = 62500 / 2^11 = 30, so  30 is greater than 16 so LDO = 1

What am I going wrong?

LoRaTracker, 2018-03-13:
> Section 4.1.1.6 says the LDO bit should be set if the symbol rate exceeds 16

Section 4.1.1.6 actually says;

"The bit LowDataRateOptimize increases the robustness of the LoRa link at these low effective data rates.Its use is mandated when the symbol duration exceeds 16ms."

The rate, symbols per second, and symbol time, ms, are different. 

My own LoRa code deals with the calculation automatically when you set the LoRa modem up, you can see how its done if you look at LoRa3.h in my library files here;

https://github.com/LoRaTracker/LoRaTracker-Library

torntrousers, 2018-03-13:
So that would be here: https://github.com/LoRaTracker/LoRaTracker-Library/blob/master/LoRa3.h?utf8=%E2%9C%93#L188-L195

So is that saying its: ( 1000 / ( bw / 2 ^ sf) ) > 16?

so 

when SF is 11 and BW is 20.8k then 1000  / (20800 / 2^11) = 98 > 16 so LDO = 1
when SF is 11 and BW is 62.5k then 1000  / (62500 / 2^11) = 32 > 16 so LDO = 1
when SF is 11 and BW is 500k then 1000  / (500000 / 2^11) = 4 < 16 so LDO = 0
when SF is 8 and BW is 62.5k then 1000  / (62500 / 2^11) = 4> so LDO = 0

Does that look ok?

LoRaTracker, 2018-03-13:
Download Semtechs 'LoRa Calculator' and  check .......................

torntrousers, 2018-03-13:
looks like you must register to get that and it then says it takes 24 hours ...

torntrousers, 2018-03-13:
AFICT the code is correct as it is now. I've submitted a pull request - https://github.com/sandeepmistry/arduino-LoRa/pull/121

sandeepmistry, 2018-05-05:
Closed via #121.

closed 2018-05-05 by sandeepmistry