sandeepmistry/arduino-LoRa #86: Experiencing a Race Condition

webhat, 2018-01-16:
I've been bothered by a bug which I found, which I think is a race condition, where the Arduino is reading the RFM95 quicker than the result is available.

The code in [src/Lora.cpp#506-520](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L506-L520) below doesn't wait long enough between chip select and SPI Transaction, meaning that the example code [src/LoRa.cpp#L86](https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L86) results is an `0x00` rather than the expected `0x12`. Introducing a delay, 10ms, solves this for me.

```c++
uint8_t LoRaClass::singleTransfer(uint8_t address, uint8_t value)
{
  uint8_t response;

  digitalWrite(_ss, LOW);
// insert delay here
  SPI.beginTransaction(_spiSettings);
  SPI.transfer(address);
  response = SPI.transfer(value);
  SPI.endTransaction();

  digitalWrite(_ss, HIGH);

  return response;
}
```

For clarity I am only using code from this repository, no additional custom code.

sandeepmistry, 2018-01-19:
@webhat interesting, what hardware are you using for:

- the Arduino or Arduino equivalent
- the Semtech side

webhat, 2018-01-19:
I'm using a Geekcreit UNO, with a RFM95 board.

sandeepmistry, 2018-01-22:
@webhat what are you using as the logic level converter?

Have you tried lowering the SPI frequency as per the read me? https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#set-spi-frequency

sandeepmistry, 2018-02-16:
Closing for now due to lack of activity.

closed 2018-02-16 by sandeepmistry