sandeepmistry/arduino-LoRa #87: Setting the broadcast frequency

31warren, 2018-01-17:
Please advise how to set a default broadcast frequency e.g. 869.525

I am using this library with a MKR1300 board.

webhat, 2018-01-17:
You can pass the frequency to `begin` as a `long`, so that would mean something like `LoRa.begin(869525000)`.

```c++
int LoRaClass::begin(long frequency)
```

31warren, 2018-01-17:
Thanks,
Just to confirm, in the LoRaSender example could I use

  if (!LoRa.begin(869.525000)) {
    Serial.println("Starting LoRa failed!");
    while (1);

webhat, 2018-01-17:
No, you misunderstood. You don't want to use a `float` `869.525000`, you want to use a `long` `869525000`, without the decimal point.

```c++
if (!LoRa.begin(869525000)) { // NO decimal point
    Serial.println("Starting LoRa failed!");
    while (1);
}
```

31warren, 2018-01-18:
Thank you for your help.

closed 2018-01-18 by 31warren