sandeepmistry/arduino-LoRa #88: Issues with LoRaDuplex

r28kumar, 2018-01-18:
Hello,

I am working on my project and I am using [LoRa/GPS shield](https://www.robotshop.com/uk/lora-gps-long-range-transceiver-shield-868-mhz-eu.html?gclid=EAIaIQobChMI3d6mrZzi2AIVSJ7ACh2UzgLNEAAYAyAAEgKfu_D_BwE) with my Arduino Mega 2560. I want to set up sender and receiver for my application and I have 2 units each. I used the [LoRaDuplex](https://github.com/sandeepmistry/arduino-LoRa/tree/master/examples/LoRaDuplex) example which I downloaded with LoRa library. I made a slight modification to the code (for the pins) and put the same application on 2 units to set up 2-way communication on both units.
When I run the units I can see in my serial ports that Data being transmitted from 1 unit is received by other and vice versa. However, after some time (ranging from a couple of seconds to 3-4 minutes, not fixed duration) or after 5-6 (or more) TX/RX both units do not receive anything. I can see both units are trying to send the data but nothing is received. Did anyone experience similar issues?

For my application, I am using the example code as a base and added very simple logic to transmit my data. What's puzzling me is that after some time that Data that I send isn't same what I receive on the other end...! I stripped down all of my application to bare bones so I am only dealing with TX/RX part to eliminate any other thing in my code which could cause the data reception to be different than what was transmitted. However, I still had the same issue and that's when I started experimenting with LoRaDuplex example a bit more and realized that the given code doesn't function well.

Any help on this issue would be greatly appreciated. 

Thank you.
Rav

sandeepmistry, 2018-01-19:
@r28kumar could you please share the EXACT sketches you are using? Is it just the [LoRaDuplex](https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplex/LoRaDuplex.ino) from the repo or did you make changes? Please also provide the output from the Serial monitor as well.

How are you powering the boards?

r28kumar, 2018-01-19:
Hello Sandeep,

Thank you for responding. Following you will find the sketch that I used both units. The LoRaDuplex code is from the repo but I commented out the pin definitions and default override in the code as I used the default pin configuration declared in LoRa.h file (downloaded with LoRa library). I have also attached 2 screenshots of Serial monitors and you can clearly observe that reception stopped within 5 minutes or so. The difference in the message ids (in the screenshots) is due to the fact that both units were not powered up at the same instance.

The units (Arduino Mega with LoRa/GPS shield mounted) were USB powered through my Laptop.

Thank you,
Rav

[LoRaDuplex.txt](https://github.com/sandeepmistry/arduino-LoRa/files/1645054/LoRaDuplex.txt)

[image]

[image]

sandeepmistry, 2018-01-22:
Are both boards sending the messages around the same time?

If so, this LoRa is half-duplex and broadcast, each board will not be able to receive the others message. 

Would you be able to change the random time in the sketch from every 2 - 3 seconds to, 20 - 30 seconds or higher?

KE6MTO, 2018-02-05:
Just to make sure, you have the antenna's attached to each node? Is the reg on the Arduino getting hot, maybe overtaxing the regulator? For what it's worth I had a similar issue with a Particle Proton (just stops receiving but more at hours vs minutes) but assumed it was something on that platform as I don't see it on the MCU's I've tried with this library.

r28kumar, 2018-02-05:
Hello @sandeepmistry,

I had set the time up to 1 minute. The communication stops but now it takes longer to arrive at the problematic state as described in my previous update. I power the units at different times to ensure that there is no interference between TX/RX as I figured that LoRa bees are not full duplex.

@KE6MTO , Yes, the antennas are attached to each node. I have checked on both hardware sets and nothing is getting hot. 

Could it be some sort of buffer, when not cleared, causes the LoRa bees to not receive the messages properly?

Thank you,

wero1414, 2018-02-16:
Hi @r28kumar, have you tried to use an SDR or spectrum analyser to know if both nodes are still sending information? 

Just want to assure that the error is at the receive part and not in the transmit.

Black3rror, 2018-03-27:
Hi. I have the same problem. after a random time the receiver doesn't receive anything.
I've set one module as Tx and the other as Rx (by //commenting the other side for each of them).
I'm using 2*( arduino uno + LoRa-RA02(sx1278) )

kovendhanj, 2018-05-13:
Hi.i have the code for both Tx and Rx in same module. you can try this code.but while we send the data there is some delay to receive.
#include <SPI.h>
#include <LoRa.h>

char c;
void setup() {
  Serial.begin(9600);
  LoRa.begin(433E6);
  LoRa.setSpreadingFactor(10);
  LoRa.setSignalBandwidth(62.5E3);
  LoRa.crc();
}

  
   
void loop() {
            while (Serial.available())
            
            { 
            LoRa.beginPacket();
            c=Serial.read();
           LoRa.print(c);
           LoRa.endPacket();
               
            }
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
  while (LoRa.available()) {     
      Serial.print((char)LoRa.read());
    }
  }
}

gpsolarco, 2019-08-17:
I see a slightly different issue with my example. Although messages are send every 2-3 seconds, only intermediate messages are received about 30-40s apart:

[image]

Not sure what could be causing the delay. The 2 kits are right next to each other.

s19sara, 2019-10-18:
Hi,
I have used Ra-02 LoRa module , sending data from sender to receiver module is working fine. How I send acknowledgement from receiver to sender module?

closed 2020-01-28 by morganrallen

AjayJassal55, 2020-11-02:
> Hi.i have the code for both Tx and Rx in same module. you can try this code.but while we send the data there is some delay to receive.
> #include <SPI.h>
> #include <LoRa.h>
>
> char c;
> void setup() { [...]

this program is also not working as full duplex

IoTThinks, 2020-11-03:
SX1278 is not a Full Duplex chip. 
Either TX or RX 1 packet at a time.

By the specification.