sandeepmistry/arduino-LoRa #90: How to configurate LoRa for best distance?

MauriPastorini, 2018-01-23:
Hi, I was wondering how can we configurate the LoRa with the library for the best distance? Something like changing the data rate, baud rate and modulation. And how to request which values had been set for confirmation.

Thank you

morganrallen, 2018-01-23:
First step is getting a better antenna. The stub and wires these usually come with are rubbish. I used a proper 800-850MHz and that increased my range significantly (828 meters so far). In that same test I had the following settings.

Bandwidth: 250E3 (less might have been better)
Frequency: 915E6 (866E6 *would* have been better, duh, antenna)
Spread Factor: 9
Preamble: 8
C/R Denominator: 8

I tried a follow up test but it didn't work at all due to not taking transmission time into account. So something to keep in mind, the longer your range, the longer you need to wait between TXes. Here's a handy spreadsheet to calculate On Air Time. 
https://docs.google.com/spreadsheets/d/1voGAtQAjC1qBmaVuP1ApNKs1ekgUjavHuVQIXyYSvNc/edit#gid=0

morganrallen, 2018-01-23:
Oh, CRC was disabled too.

Hoel, 2018-01-25:
Hi,
What is the default BW when not explicitely set in the app? I did not find any reference to setSignalBandwidth in the init, so i wonder what the library default to.
Thanks

morganrallen, 2018-01-25:
Looks like 125kHz is the default.

http://www.mouser.com/ds/2/761/sx1276-944191.pdf Page 113

Hoel, 2018-01-26:
OK Bw is let to chip default setting.Thanks.

MauriPastorini, 2018-01-26:
Excellent morganrallen, many thanks!
Sorry about the really basic questions but besides the antenna, 

Bandwith: If reduce the bandwith at minimum, lets say 7.8E3, am I going to  have better distance?
Spread factor: For distance which is better? More or less? Between the 6 and 12.
Preamble: For distance which is better? More or less? Between the 6 and 65535
Coding Rate: For distance which is better? More or less? Between the 5 and 8
CRC: For distance which is better? Enable or disable?

If anybody could answer this parameters for distance would be awesome

morganrallen, 2018-01-26:
Spread factor is going to be the one that determines distances the most (bandwidth second?). Higher = longer range AND longer TX time. All the other params will mostly affect transmission time.

sandeepmistry, 2018-02-05:
Thanks @morganrallen!

I'm closing as answered for now.

closed 2018-02-05 by sandeepmistry

saedelman, 2018-02-13:
@MauriPastorini - the answers provided are not quite complete. Please download and play with the Semtech "Lora Calculator Tool" that you can download from their website (Windows only). It will clearly show you how the spreading factor (SF), Bandwidth (BW) and coding rate (CR) affect time-on-air (affects power consumption) and receiver sensitivity (affects distance). If you don't care about time on air/power consumption, set the bandwidth to a minimum and SF to maximum. You'll get an incredible link budgets with line of sight communication of 100km or more. Check out Andreas Spiess' Youtube videos on Lora for a test he did over very large distances. The antenna had very little impact.

saedelman, 2018-02-13:
@MauriPastorini - One additional detail that is often overlooked - the lower the bandwidth, the more stable of a clock you need driving the Semtech chipset. For example, a number of Hope RF modules use a crystal oscillator with 10ppm drift, which means you cannot select bandwidths of less than 41.7 kHz (31.25 kHz may work) or you will not be able to demodulate any packets. Note that operation in high G environments will affect clock drift as well.  In general, if you are designing strictly based on link budget (receiver sensitivity), choose the higher bandwidths (more immune to clock drift). For example, SF 10, BW 500 kHz and CR 4/5 yields a receiver sensitivity of -126 dBm with an on-air time of approx. 100ms for a 32-byte packet (6 byte preamble) but SF 9, BW 250 kHz and CR 4/5 also yields a receiver sensitivity of -126dBm with a similar on-air time (around 109ms). The latter allows for a maximum crystal drift of 72ppm whereas the 500kHz bandwidth will allow 144ppm drift. Therefore, for the same link budget, I would choose the wider bandwidth (500kHz) and and corresponding larger spreading factor (10) (less prone to interference than SF 9). Hope this helps.

morganrallen, 2018-02-13:
Thanks for this more detailed response, this would be extremely useful as a footnote in the docs if you feel like writing up a "Quick Guide to Range".

saedelman, 2018-02-13:
@morganrallen - one of the best presentations on LoRa that I have come across is from Matt Knight at Bastille Networks - [Reversing Lora](http://bit.ly/2EpPyix). Skip to "Introducing LoRa". 

A couple more comments on this, you recommended in a previous post to set the C/R to 8 (ref. C/R Denominator: 8), this effectively sends your message twice and doubles your on-air-time (and doubles current-consumption). If it is important to have your message received at the other end within a certain period of time, consider sending data more frequently. E.g. if your receiver needs to see data ever 30 seconds, send data every 15 seconds. This has the benefit of providing more up-to-date data (rather than send two copies of the same data with C/R 8) and you space the packets in time (possibly avoiding temporal interference). 

In the event you are using LoRa modulation outside of a gateway configuration (e.g. communication from node to node in a broadcast to all configuration), the longer the on-air-time the more collisions you will encounter (multiple nodes talking at the same time). I personally don't use any C/R above 6 for that reason alone. 

There have been comments about whether CRC checks should be turned on. These should always be turned on even when you have an application-level checksum. What is the sense of having the LoRa modem send you a garbage packet? In addition, the cost of sending you the garbage packet is not zero. In many cases, the LoRa modem is connected by UART (or I2C) and at speeds of 57600, the cost of sending you a bad packet can be tens of milliseconds and then you have to instruct the modem to restart the listening process and during this whole process you are missing out on receiving other packets. Turn it on. Always.

saedelman, 2018-02-13:
@morganrallen - Yes, excellent point.

universam1, 2018-02-14:
@saedelman This is really valuable information! Would like to encourage you to post this findings to the FAQ!

halukmy, 2018-02-15:
please update readme and faq for better distance settings

saedelman, 2018-02-15:
The following table is based on midband frequency of 915MHz (North America) and sender and receiver isotropic antenna gains of 1 (e.g. a whip antenna without any gain)

Distance - SF/BW
10km - 7/500
20km - 8/500
30km - 9/500
40km - 10/500
60km - 11/500
80km - 12/500
100km - 11/250
125km - 12/250

Although there are other possible combinations of SF/BW possible to achieve the indicated distances, the above table is based on minimizing time-on-air (and therefore minimal power consumption and maximum data rate). The above table is strictly based on free space path loss and therefore distances must be line-of-sight. Note that it is not just a matter of the tips of your send/receive antenna seeing each other without obstructions, there is a so-called Fresnel zone that needs to be entirely clear of obstructions. This zone is like a "blimp" suspended between the tips of the transmit/receive antennas. The maximum diameter of the "blimp" is a function of distance. There are many good tools out there to plot the Fresnel zones between two geographic points - see https://www.loxcel.com/3d-fresnel-zone (not free).

tarun-kumar-m, 2018-11-20:
> The following table is based on midband frequency of 915MHz (North America - you'll get better results when using 868MHz) and sender and receiver isotropic antenna gains of 1 (e.g. a whip antenna without any gain)
> Distance - SF/BW
Please can you provide reference for theses calculation as I want to experiment with different SF/BW and different ISM band (868 MHz) to understand possible communication range.

enriqueromero, 2019-10-17:
Hi guys,
So I get that CRC should be on. My question is more max transmitting power out of a LoRa module.
In the file LoRa.cpp file the max power is set to:
setTxPower(17); is this really the max?
Also, is anything to be done with following in the Lora.h file:

#define PA_OUTPUT_RFO_PIN               0 
#define PA_OUTPUT_PA_BOOST_PIN     1

Note: in my case, I'm only transmitting 32 bytes but I need max range-reliability(who doesn't) :)
Thanks in advance.
          ER

firzov, 2020-04-02:
Hi everyone, I was hoping to get a little help, me and a colleague of mine are tryiing to set up a Lora transmitter and reciever. 
The transmitter that will send data in form of flow are the type Feather 32u4 RFM95 LoRa, Adafruit, and in the recieving end we have a raspberry pi 3 with Lora/GPS for pi V1.4. 

We live in sweden and are using 868mhz. 

We seem to get distances working on around 1km free line of sight, but not so much further. 
In one end we have a "normal" LTE antenna for a router (which is capable of transmitting on 868mhz but not sure of the dbi, and on the Pi side we have Poynting Xpol-2-5G LTE Mimo 11Dbi 698-3800 MHz. 

We have problem when there are the slightest obstacle, can someone recommend the best BW/SF, etc that we can just start from ? We have tried somethings but arent completely happy with result.

dnyaneshvars, 2021-02-23:
is there anybody out there who has successfully managed to get 1km distance for Lora transmission in (433mhz India)?

IoTThinks, 2021-02-23:
In open space, yes.
Btw, few people will trust others’ test results.

> is there anybody out there who has successfully managed to get 1km distance for Lora transmission in (433mhz India)?

rmahajan9, 2021-08-04:
> is there anybody out there who has successfully managed to get 1km distance for Lora transmission in (433mhz India)?

Hi there, even I am struggling to get any more than 500-600meters coverage from my LoRa nodes in open fields. I am using proper ducky antennas yet not of much success. Any suggestions? My Lora module is 433Mhz SX1278 fed with 3.2v 2000mah Lipo & controlled using a NodeMCU. 
Any suggestions to increase the range between the node will be of great help. 

[image]
[image]

IoTThinks, 2021-08-05:
You should create another issue next time.
Btw, post your RSSI / SNR results at 1 meter and 5 meters with strictly no obstacle between two nodes.

Kongduino, 2021-08-05:
I reached 2 km with 2 crappy TTGO LoRa v1, no LoS.
I reached 10 km with 1 crappy TTGO LoRa v1 at home with a Yagi, and 1 TTGO T-Beam with a 15-cm antenna, no LoS. If you can't get 1 km there's definitely a problem...

https://kongduino.wordpress.com/2021/06/16/so-i-bought-a-yagi/

rmahajan9, 2021-08-05:
at 1 mtr distance RSSI value was -90 & at 5mtrs it was -114 (this was when tested indoors but without any obstacle between the nodes).

IoTThinks, 2021-08-06:
> at 1 mtr distance RSSI value was -90 & at 5mtrs it was -114 (this was when tested indoors but without any obstacle between the nodes).

Very likely antenna issue. 
At 1m, the RSSI should be around -50 to -60.
At 5m, the RSSI should be around -70 to -80.

You can make your own bronze wire antenna to verify.

433Mhz
299792458 / (433000000 * 4) * 100 = 17.31cm
0.95 * 17.31 = 16.445cm (Antenna with bare copper)

ashishmurikingal, 2021-11-11:
its the same case with me, did you find any solution?

rmahajan9, 2021-11-12:
Unfortunately no, I didnt find any solution to it. from lots of forums I
learnt that people who could get even 1 or 2Km range was due to the fact
that they mounted the LORA Transmitter at a high rise building.  With such
small power or 3.3v or 5v you cant expect getting more than 1000m range
unless u use a highly directional tx and rx antennas.

Regards,
Rajeev Mahajan
93502 30292

IoTThinks, 2021-11-12:
> at 1 mtr distance RSSI value was -90 & at 5mtrs it was -114 (this was when tested indoors but without any obstacle between the nodes). 
> 
> 

Your antenna is likely wrong tuned for your frequency.
-90 RSSI at 1 meter is equivalent to no antenna at all.

However, more than 1km is usually with no obstable at between. High position is always a plus to avoid signal to bounce to the ground.

Kongduino, 2021-11-12:
Oh yee of little faith. As I said in a previous answer, I got 2 km, no LOS, on crappy TTGO devices. My transceiver was 15 meters up, ie not high up, and in a neighbourhood with lots of high-rise buildings getting in the way. I got 10 km with better antennas, but still between TTGO devices (a T-Beam and a TTGO LoRa OLED). A couple of days ago I got 1.3 km at RSSI -94 ~ -95 between 2 devices, with the new STM32WLE5CC chip, that had small, default antennas. I could have gone much further out, but didn't see the point: it was obvious I could have reached a couple of km.

> small power or 3.3v or 5v you can't expect getting more than 1,000m range

IoTThinks, 2021-11-12:
@Kongduino Quite impresive. 
Would you mind sharing the antenna you use?
Thanks a lot.

Kongduino, 2021-11-12:
@IoTThinks 

I use, for the receivers, that kind of antennas – this Taobao shop is my go-to. I have most of them, in 433, 868 and 915. They're pretty decent. I put them through my NanoVNA to decide what the best frequency for each is – sometimes quite a few MHz away from the nominal frequency – and use that.

[image]

Between two of those, I get 3/4 km, easy, sometimes more – the weather also influences results. But OTOH they spread wide and large: I really get 360° coverage, even with the transceiver pointing west, and the east side being hidden by my building.

Between one of those and a Yagi, well, at least in the 433-470 range, 10 km is nowhere near my limit I believe (and I need to do tests at 430 MHz, as that frequency is quiet, at least here). I don't really get 360° – at least not beyond 2 km , but that's quite good 360° coverage for a directional antenna. My next target is 24 km, a mountain range that faces my windows, far far far away.

[image]

In the 868 range, I'm a little less enthusiastic, but I think I can beat my record of 7.5 km. I just need a little work on the settings and SF/BW combination. And maybe another Yagi... :-)

[image]

> @Kongduino Quite impressive. Would you mind sharing the antenna you use? Thanks a lot.

IoTThinks, 2021-11-12:
@Kongduino Thanks a lot for your sharing. I have a NanoVNA too.
Let me check check my antenna and see how far I can push my current antenna.

Kongduino, 2021-11-12:
@IoTThinks 
The problem with the NanoVNA is that it doesn't go beyond 900 MHz – which sucks for 915 antennas. I'm looking for another VNA to check on 915 MHz and 2.4 GHz antennas...

IoTThinks, 2021-11-12:
@Kongduino By right, my Nano VNA (v2.2 Plus) can reach 3Ghz.
https://nanorfe.com/nanovna-v2.html#buying

tomicooler, 2022-01-10:
> ```c
> TURN OFF LNA FOR TRANSMIT
> ```

Hi,

in my project I use HopeRF RFM95W module with a very cheap 2 dBi antenna both on the sender and receiver side. I could reach 700 meters in rural area (the receiver was not placed high and there was some up and downhills). In the city I managed to get 200 meters only. I need to achieve 350 meters for my project.

Currently, I use the following settings:

```
  LoRa.begin(868E6);
  LoRa.setSpreadingFactor(12);
  LoRa.setSignalBandwidth(125E3);
  LoRa.setCodingRate4(8);
  LoRa.setTxPower(20);
  LoRa.enableCrc();
```

I tried with spreading factor 11 and 62.5E3 bandwidth too, the SF12 was probably better. Since the sender and receiver is in a fixed location I think I could use directional antennas or try higher gain antennas. This is my first LoRa project so a lot of reading needs to be done on my part before buying an expensive antenna set.

What does the `LoRa.writeRegister(REG_LNA, 00); // TURN OFF LNA FOR TRANSMIT` and `AGC = 0x0c` mean?

https://drive.google.com/file/d/18i8pvsvV6gIV2blT0ikxekerrOwzd2EK/view?usp=sharing
https://drive.google.com/file/d/1sa6JZH7Er81Urh6pAyaDeWC1tfLRU_d8/view?usp=sharing

BTW LoRa.begin() sets these values:
```
  // set LNA boost
  writeRegister(REG_LNA, readRegister(REG_LNA) | 0x03);

  // set auto AGC
  writeRegister(REG_MODEM_CONFIG_3, 0x04);
```

0x04   0100
0x0c   1100

0x0c sets the LowDataRateOptimize flag too? Would that help?

I just noticed the issue is closed already, but I have a question about the quoted code snippet, I didn't want to open a separate issue for that.

s17534, 2024-04-15:
>   uint8_t BW = 0x48, SF = 0xc4, AGC = 0x0c; // SF 12 BW 125KHz AGC
> ```

I think you are setting 31.25 kHz BW instead of 125 kHz

0x48 means 0100 1000 and as per sx127x datasheet:

[image]

7 - 4 on address 0x1d are used to set BW,  125kHz (default) is 0111 not 0100

btw. why using private functions instead of those public ones for example LoRaClass::setSpreadingFactor(int) etc?

taloot, 2024-08-21:
> Unfortunately no, I didnt find any solution to it. from lots of forums I learnt that people who could get even 1 or 2Km range was due to the fact that they mounted the LORA Transmitter at a high rise building. With such small power or 3.3v or 5v you cant expect getting more than 1000m range unless u use a highly directional tx and rx antennas. I suppose low range issue could also be due to the sub-quality LoRA modules supplied to Indian mkt from China. Regards,
> […](#)

becouse u are canjoos. u buy less known brand, same as what i did,, and i found that u have to   LoRa.setDio2RfSwitch(true);
otherwise it will be very weak