sandeepmistry/arduino-LoRa #91: Starting LoRa failed! Version = FF

royheuer, 2018-01-24:
Good day

I am using a LoRa32U4 device from BSFrance and I get a Starting LoRa failed! response. I have looked through the various issues and have attempted to get the version using:

 uint8_t version = readRegister(REG_VERSION);
  if (version != 0x12) {
    return Serial.println(version, HEX);
  }

The response is FF.

The pin settings are as follows:

//LoRa32u4 ports
#define SS      8
#define RST     4
#define DI0     7
#define BAND    433E6

royheuer, 2018-01-24:
The different versions of the board has different SS pin.  My version has SS pin as 1

closed 2018-01-24 by royheuer

aahondema, 2018-01-24:
I use the same board: these are my PIN-settings
`lmic_pinmap pins = {
  .nss = 10,			// Connected to pin D10
  .rxtx = 0, 			// For placeholder only, Do not connected on RFM92/RFM95
  .rst = 9,  			// Needed on RFM92/RFM95? (probably not)
  .dio = {2, 6, 7},		// Specify pin numbers for DIO0, 1, 2 connected to D4, D5, D7 
};`