sandeepmistry/arduino-LoRa #93: Example not working

srpn, 2018-01-29:
I started with two lora modules (Ai-thinker Ra02) with Arduino Uno and tried peer to peer connection based on the examples (LoRaReceiver and LoRaSender). The sender seems to work, however the receiver is not receiving any packets. 
There is no other error (initialization seems fine). When I tried LoRaReceiverCallback, it kept on printing empty packet [Received packet '' with RSSI -164]

Any pointers would be appreciated.

sandeepmistry, 2018-01-31:
Please see https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power

closed 2018-01-31 by sandeepmistry

halukmy, 2018-02-01:
wow so useful, thanks @sandeepmistry

srpn, 2018-02-05:
@sandeepmistry
I do not think this is an issue with the tx power. I was prototyping and the tx & rx modules were only a couple of feet apart. 
Also as i said, LoRaReceiverCallback kept printing 'empty' packets. Does it mean anything?

Would appreciate if you could reopen this issue.
[Anyways i tried tweaking the power options, but still same effect]

parvgupta001, 2020-04-10:
the problem i was facing that default transmitting frequency of code was not matching the one supported by my lora module so i changed from 915-433 hz and it worked.frequency - frequency in Hz (433E6, 866E6, 915E6)
for more details visit-https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power

GeospaceMD, 2022-05-20:
@srpn I am facing the exact same issue to you right now using two arduino mkr wan 1300 modules. Did you ever get this figured out?