sandeepmistry/arduino-LoRa #98: I have two LoRa Dragino modules and I can't get them to communicate with each other using this library

GurenMarkV, 2018-02-01:
I like your library as it is easy to configure all the settings to try and optimize distance. I have these two Arduino shields and can't get them to talk to each other using the Duplex or the send/receive code. Is there modifications that need to be made to the duplex code? I just uploaded the sample as is on both devices.

halukmy, 2018-02-02:
you can n0t go and buy a dorji sx1278 f module!

sandeepmistry, 2018-02-05:
Make sure to update the frequency passed into begin to match what your h/w is tuned for.

KE6MTO, 2018-02-05:
All connections double checked including the DI0 line?

GurenMarkV, 2018-02-07:
I will try the frequency change later on. Might take me a few weeks to get back to you. 

> All connections double checked including the DI0 line?

It's a shield so all the connections were as they should be. 
Using the RadioHead library I can get 20m. I wanted to try using this one to get 3km after some optimization on the code.

sandeepmistry, 2018-02-16:
Ok, please try it out and re-open this issue if you still have problems.

closed 2018-02-16 by sandeepmistry

pallavisingh1998, 2019-02-22:
on using radiohead library range between two lora shields is very less. how to increase the range upto 1 km?