sandeepmistry/arduino-LoRa #100: how can i up sender value than 255?

halukmy, 2018-02-04:
i just tried to 1234 sender value but it says only 210 after change from hex,

so if i got 1000 node how can i up it?

thanks

sandeepmistry, 2018-02-05:
@halukmy you'll have to split the write up byte by byte, as per the Stream interface

```arduino
  // send packet
  LoRa.beginPacket();
  LoRa.write(0x12);
  LoRa.write(0x34);
  LoRa.endPacket();
```

closed 2018-02-05 by sandeepmistry

halukmy, 2018-02-05:
@sandeepmistry 

byte destination1= 0xBB;     // address of this device
byte destination2 = 0xFF;      // destination to send to

how can i split parse like that?

  LoRa.write(destination1);              // add destination address
  LoRa.write(destination2);              // add destination address