sandeepmistry/arduino-LoRa #101: Call for co-maintainers!

sandeepmistry, 2018-02-05:
Would anyone be willing to step up and become a co-maintainer of this project?

cc/ @morganrallen @universam1 @annikW

sandeepmistry, 2018-02-05:
@Rotzbua :)

Rotzbua, 2018-02-05:
@sandeepmistry Thanks :) but I already spent my rare free time with maintaining the mfrc522 lib..
I hope you will find some co-maintainer, it is always sad to see good project dead.

universam1, 2018-02-05:
Thanks for the call!
I am working on a bigger LoRa project this year so this might come handy, but I am limited also time wise. What would you expect to collaborate?

sandeepmistry, 2018-02-05:
@universam1 nothing too intense, I'll grant you write access to the repo + add your name inside library.properties, and maybe once a week you can help respond to open issues, review pull requests, or submit pull requests for bug fixes.

Again, this is whenever you have some spare time ...

Rotzbua, 2018-02-05:
This project uses already travis as CI system to check compilation of the code which makes maintaining a lot easier. 👍 

I recommend some templates for issues and pull requests. This also safes a lot of time. Well sometimes you have to be rude and close issues if beginners ignore the template ... your time is more important than the time of the beginner ;)

Example templates:
https://github.com/miguelbalboa/rfid/tree/master/.github

universam1, 2018-02-05:
@sandeepmistry sure I can help out that way to keep the project moving. 
You don't need to mention me since it's really your effort that made this library. In doubt I would always ping you anyway.

sabas1080, 2018-02-05:
@wero1414

wero1414, 2018-02-06:
I could be co-maintainer i have some experience working with the SX1276 directly by SPI and so, may be usefull

morganrallen, 2018-02-11:
Always happy to help out. As I work on my C driver I'll always be testing against this library anyhow, so count on me fixing anything I run into along the way.

werty37, 2018-02-11:
@sandeepmistry What is the role of a maintainer? Can you please list it out?

werty37, 2018-02-11:
@sandeepmistry I can spare some time every week for this.

sandeepmistry, 2018-02-11:
I've invited @universam1 @wero1414 and @morganrallen for write access to this repo, thank you for volunteering! Please let me know if you would like your names (and emails) added to the maintainer field of the `library.properties` file.

Please avoid directly pushing code to master, instead open a pull request for any code changes. I will do the same from now on as well. Then once one of us reviews the pull request we can merge. We should all have consensus on API changes, bug fixes and issue are a bit more relaxed.

@werty37 thank you for your interest, for now if you can monitor issues on this repo and help people out with comments. If this goes well, later on you can get write access to close tickets.

closed 2018-02-11 by sandeepmistry