sandeepmistry/arduino-LoRa #102: Kit LoRa radio Sx1276 (not working)

samueldiazlopez, 2018-02-13:
A few days ago I bought this kit:
http://tienda.bricogeek.com/lora/1072-kit-lora-radio-sx1276-con-antena-433-mhz.html

The seller says it is compatible with your library but I can't find a way to connect it to the arduino so that it works.
http://wiki.dfrobot.com.cn/images/0/0e/WL800T.png

Can someone tell me how to connect it to the arduino?

morganrallen, 2018-02-13:
You'll need to provide more details, "Not working" doesn't provide much to start debugging with.

* What example code are you using?
* HOW do you know it's not working? (Error?)
* What steps to debug have you taken so far?

samueldiazlopez, 2018-02-14:
The problem is that it is not connected to the arduino. :(

morganrallen, 2018-02-14:
Not sure what we can do to help.
I don't have access to one of those boards, it's pin out **is not** SPI, so this driver cannot even possibly communicate with it. In fact, from that pin out, I wonder, is the board itself and Arduino?

But... without providing more info ("not connected" is not good info) there is little help we can provide.

closed 2018-02-14 by morganrallen