sandeepmistry/arduino-LoRa #104: LoRa Packet Payload length

tharindu147, 2018-02-13:
Please Some one tell me how to set the LoRa Packet Payload length. I need to set it to 4 bytes.

morganrallen, 2018-02-13:
**Packet length**
By default LoRa radios transmit in "explicit header mode", meaning the header is sent with payload length and forward error correction set. So I think, this is something you don't need to worry about. This library also provides method to specify exact length, if say, you want to only send the first 4 bytes of a longer array.

```
LoRa.beginPacket(); // no param, explicit is default
LoRa.write(&longArray, 4);
LoRa.end();
```

For more info, see
Semtech SX1276/77/78 WIRELESS & SENSING PRELIMINARY DATASHEET
4.1.1.6. LoRa(TM) Packet Structure

**Dual Radio**
Without digging in a ton, I don't think the library is setup to handle multiple devices but off the top of my head you might be able to get something working. If the devices are on the same SPI bus, before sending you could call `LoRa.setPins` to change the `SS` pin to the target device. The biggest issue I see is onReceive interrupts could get mangled in the process. 

(in the future, it could be extremely helpful to future visitors to ask separate questions like these in different Issues, thanks).

sandeepmistry, 2018-02-16:
@morganrallen regarding multiple radios the user could add a `LoRaClass LoRa1;` in their sketch or library. The only issue with this is getting the interrupt RX working if `LoRa1.onReceive(...)` is called as `void LoRa::onDio0Rise();` is currently static and only expects one `LoRaClass` instance.

sandeepmistry, 2018-02-16:
I guess another key question is why would someone what to use multiple radios on a single board?

morganrallen, 2018-02-16:
This was discussed at the Oakland Disaster Radio meeting last week. In building a large scale mesh network, multiple radios could be desirable to talk with closer APs at higher speeds while still being able to communicate with the larger network.

tharindu147, 2018-02-16:
Thank you for reply.
Reason for using two radios is I need to use 920MHz and 433MHz radios to receive in both bands.  

(in future I will ask separate question in different issues).
Thank you.

sandeepmistry, 2018-02-26:
Ok, I'm going to close this now. Please open a new issue to track multiple LoRa radios on one MCU.

closed 2018-02-26 by sandeepmistry