sandeepmistry/arduino-LoRa #107: hello, I'm not receiving the data on arduino with LoRa with the same code,     i could see serial monitor for sender like:   Sending packet 0 Sending packet 1   but for reciever monitor output is like; Starting LoRa failed!

aniketmad, 2018-02-21:
hello, I'm not receiving the data on arduino with LoRa with the same code,     i could see serial monitor for sender like:   Sending packet 0 Sending packet 1   but for reciever monitor output is like; Starting LoRa failed!

aniketmad, 2018-02-21:
[image]
[image]

sandeepmistry, 2018-02-26:
@aniketmad please let us know what hardware you are using. As well as what you've done for far to debug the problem.

aniketmad, 2018-03-31:
We are using Lora SX1276.We are getting garbage value while receiving data.

aniketmad, 2018-04-01:
or maximum times we get:  "starting lora failed!" at receiver serial monitor

morganrallen, 2018-04-06:
If you try LoraSender firmware on the receiver does it work?
Or do you get "starting lora failed"?

aniketmad, 2018-04-28:
do you mean swiping sender and receiver??
yes it does work for this.

now i'm not starting lora failed.

NOW THE PROBLEM IS(OUTPUT):
LoRa Receiver
Received packet '' with RSSI -163    **# THIS COMES WHEN I remove NSS pin

**
Received packet '' with RSSI -159
Received packet 'Ia	⸮a	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮Ia	⸮0Ia	⸮0⸮	Ia	⸮a	Ia	⸮Ia	⸮0⸮$⸮

aniketmad, 2018-04-28:
i get either only **Lora Receiver** as heading and nothing below

or else when  i remove NSS pin i receive the abbove data with that garbage value.

please help me soon.

it's urgent

conradholt, 2018-07-18:
I had the same problem, but I updated the standalone firmware for MKR WAN 1300, (that basically acts as a bootloader for the stm32) using the example MKRWANFWupdate_standalone inside the MKRWAN library and solved this issue.

sandeepmistry, 2018-08-19:
Closing for now due to lack of activity.

closed 2018-08-19 by sandeepmistry

shivampandey3611, 2018-10-14:
sir I am sending values temprature and humidity but it is not able diffrentiate the values as well as not printing values on lcd print please suggest us some code.for devlopment purpose

deanpretorius, 2019-07-30:
Hi. Was there any progress with this issue? I have the same issue and am not sure how to proceed with solving it.

ClaytonMendes, 2019-11-28:
I'm sorry for my english, I'm having the same problem.
Any solutions to this problem?

pkrishnavamsi0709, 2023-03-06:
i am having the same issue can anybody help me out

pkrishnavamsi0709, 2023-03-06:
i am getting some unkown data

pkrishnavamsi0709, 2023-03-06:
> We are using Lora SX1276.We are getting garbage value while receiving data.

did you get any solution this problem I am also getting the same issue

Aishwaryakanna, 2023-04-12:
I am having same probelm with Lora-e220-900T22D

morganrallen, 2023-04-12:
Lora-e220-900T22D is based on the LLCC68 chip, which is not supported by this driver.

pkrishnavamsi0709, 2023-04-14:
> I am having same probelm with Lora-e220-900T22D

My problem is solved . In my condition firstly I am using esp for transmitting from lora and Arduino Uno for receiving later I changed both transmitter and receiver to Arduino Uno then it worked

Srivardhan06, 2024-05-14:
same problem getting garbage values